waveshareteam/WM8960-Audio-HAT #1: Connecting Audio Board to Pi

mintu19, 2019-04-08:
I am trying to connect Wm8960 audio board to raspberry pi (not hat).
I have installed drivers and connected module. But when i use i2cdetect, i get no address of connected board.

hnwangkg-ezio, 2019-04-09:
First you have to check if there is a problem with your hardware connection and then turn on the I2C kernel. Then run
I2cdetect -y 1
`
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- 1a -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --            
`
address is 0x1a

hnwangkg-ezio, 2019-04-09:
First you have to check if there is a problem with your hardware connection and then turn on the I2C kernel. Then run
I2cdetect -y 1
`pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- 1a -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                  `
address is 0x1a

hnwangkg-ezio, 2019-04-09:
First you have to check if there is a problem with your hardware connection and then turn on the I2C kernel. Then run
I2cdetect -y 1

> pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- 1a -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                        
address is 0x1a

mintu19, 2019-04-09:
I have enabled i2s and i2c from boot config.

I have made connections

VCC --> PIN 1 (3.3V)
GND --> PIN 14 (GND)
SDA --> PIN 3 (I2C SDA)
SCL --> PIN 5 (I2C SCL)
CLK --> PIN 12 (GPIO18)
WS --> PIN 35 (GPIO19)
TXSDA --> PIN 40 (GPIO21)
RXSDA --> PIN 38 (GPIO20)
TXMCLK --> NC
RXMCLK --> NC

when i do i2cdetect  (even if pins are connected or not connected) i get,

0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

If I uninstall drivers and then connect i2c only and check, I get entry at 1a. But not after installing driver.

I have WM8960 audio board. I have tested it is working with linkit smart 7688. 
But with pi, i get no sound. (My guess is i2c command are not processed by drivers).

(I have connected led to pin 40 to check if there is output when i do aplay <wav file> and indeed there is. But no output on pin 38 when doing arecord.)

hnwangkg-ezio, 2019-04-09:
You can see it here：
https://www.waveshare.com/w/upload/5/54/WM8960_Audio_HAT_User_Manual_EN.pdf

mintu19, 2019-04-09:
I followed that manual and tried to apply it to audio board instead of audio hat. That is where i am getting error. I get reserved i2c at address 0x1a after installing drivers and playback and recording doesn't work.

hnwangkg-ezio, 2019-04-09:
Excuting an order：

> aplay -l

What information?

mintu19, 2019-04-09:
arecord -l and aplay-l shows codec driver wm8960 (even if card is connected or not)

mintu19, 2019-04-10:
```
pi@rpizero:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@rpizero:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@rpizero:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

hnwangkg-ezio, 2019-04-10:
The Raspberry Pi has successfully recognized the sound card, this is the sound card 0, the manual is written on the sound card 1
so you can try :

> sudo arecord -f cd -Dhw:0 | aplay -Dhw:0

hnwangkg-ezio, 2019-04-10:
if card 0:
hw:0
if card 1:
hw:1

mintu19, 2019-04-10:
same details are shown when card is not even connected.
I tried above command. No sound.

hnwangkg-ezio, 2019-04-10:
you can try :Wm8960 audio HAT

mintu19, 2019-04-10:
yes. But it is not available in my country. I found it at amazon only at 5 time price. Thats why wanted to try on board. I will try ordering hat

mintu19, 2019-04-10:
I have found the problem. The two problems were 
1. Wires were long. Using short wires did the trick
2. Using either external 3.3 V source or 5v gpio to 3.3v buck convertor

Still one issue pertains that, if once it is disconnected in between, need to restart pi to make it work again.
It draws current same value after disconnect and reconnect but won't work until restarted.

mintu19, 2019-04-10:
Current issue is solved and working directly with 3.3v rail as well but very noisy. 
Only problem is restart issue.

closed 2019-04-11 by mintu19

AndyEngland521, 2020-02-14:
@mintu19 are you talking about your I2C lines being too long or your I2S lines being too long?

mintu19, 2020-03-04:
I2S lines