waveshareteam/WM8960-Audio-HAT #3: an error occurs during installation...

DmitrySV77, 2019-05-19:
Building module:
cleaning build area.....
make -j1 KERNELRELEASE=4.19.42+ -C /lib/modules/4.19.42+/build M=/var/lib/dkms/wm8960-soundcard/1.0/build..........(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.42+ (armv6l)
Consult /var/lib/dkms/wm8960-soundcard/1.0/build/make.log for more information.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area.....
make -j1 KERNELRELEASE=4.19.42-v7+ -C /lib/modules/4.19.42-v7+/build M=/var/lib/dkms/wm8960-soundcard/1.0/build........(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.42-v7+ (armv6l)
Consult /var/lib/dkms/wm8960-soundcard/1.0/build/make.log for more information.
Created symlink /etc/systemd/system/sysinit.target.wants/wm8960-soundcard.service → /lib/systemd/system/wm8960-soundcard.service.

DmitrySV77, 2019-05-19:
pi@raspberrypi:~/WM8960-Audio-HAT $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

ed74, 2019-05-19:
I also met this problem.

Attached /var/lib/dkms/wm8960-soundcard/1.0/build/make.log
[make.log](https://github.com/waveshare/WM8960-Audio-HAT/files/3194986/make.log)

Environment:
Raspberry Pi Zero W
Raspbian: April 2019

hnwangkg-ezio, 2019-05-21:
This may be caused by the latest system of the Raspberry Pi.

DmitrySV77, 2019-05-21:
OK, what image raspbian do I need to put in order for the wm8960 module to work?

hnwangkg-ezio, 2019-05-21:
just for now:2018-11-13-raspbian-stretch-lite

hnwangkg-ezio, 2019-05-21:
https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2018-11-15/

ed74, 2019-05-21:
@hnwangkg-ezio 
> just for now:2018-11-13-raspbian-stretch-lite
> https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2018-11-15/

it doesn't work
the kernel was updated to 4.19.42-v7+ during install (install.sh)

vaghawan, 2019-05-22:
HI any update on this? Facing the same issue with my PI 3B with rasbian strech "Raspbian GNU/Linux 9 (stretch)".

819428900, 2019-05-22:
use this https://github.com/respeaker/seeed-voicecard

vaghawan, 2019-05-22:
Did that, it installs fine and the the device is shown, however the device doesn't work, neither the arecord, nor the aplay.

819428900, 2019-05-23:
The 2019-04-08-raspbian-stretch-lite I installed with the zero w, works fine.

vaghawan, 2019-05-23:
HI @819428900 it worked. But I had to reboot Pi multiple times. Thank you for the help.

DmitrySV77, 2019-05-23:
you did update and upgrade?

vaghawan, 2019-05-23:
No, I didn't. At least not the upgrade.

DmitrySV77, 2019-05-23:
but without it, get does not work ....

DmitrySV77, 2019-05-23:
Sorry. git

vaghawan, 2019-05-23:
I didn't fully understand you, I didn't use this repo, I rather used the repo of seeed-voicecard. Their installation worked after few boots. I used the Rasbian Stretch "2018-11-13-raspbian-stretch-full.img" downloaded from the rasbian.org in my PI 3B+.

DmitrySV77, 2019-05-23:
git clone https://github.com/respeaker/seeed-voicecard
Command not found

819428900, 2019-05-23:
sudo apt-get install git

DmitrySV77, 2019-05-24:
thank!  a sound has appeared.  Now I am trying to finish the pi-radio project ..... I installed the program, but there is no sound there ....

hnwangkg-ezio, 2019-05-24:
now,fix it

closed 2019-05-24 by DmitrySV77

ed74, 2019-05-24:
@hnwangkg-ezio 
It works! Thanks!

I tried to clean install 2019-04-08-raspbian-stretch-lite.img on Pi Zero W.
Driver build under 4.19.42+ and 4.19.42-v7+ kernel.

Double check device list
```
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 1: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

hnwangkg-ezio, 2019-05-24:
This seems to be working.

LiJu09, 2019-06-05:
> @hnwangkg-ezio
> It works! Thanks!
>
> I tried to clean install 2019-04-08-raspbian-stretch-lite.img on Pi Zero W.
> Driver build under 4.19.42+ and 4.19.42-v7+ kernel.
>
> Double check device list
>
> ```
> $ [...]

I try the same
Not work for me

iTommix, 2020-06-18:
For newer Kernel (in my case 4.19.66-v7+) you have to link the new Kernel-Sourcen to the installer expected 4.14.98-v7+ (i dont know why the installer expect these):

`sudo ln -s /lib/modules/4.19.66-v7+ /lib/modules/4.14.98-v7+`
or
`sudo ln -s /lib/modules/$(uname -r) /lib/modules/4.14.98-v7+`

then execute the install.sh

This worked for me.

Daveolw, 2020-09-26:
Hi,
Facing the same issue with my PI 3B with rasbian strech "Raspbian GNU/Linux 9 (stretch)
Purchased WM8960 sound card but cannot get it working,   Error! Bad return status for module build on kernel: 4.19.66-v7+ (armv7l)
End of Log file shown.
[log.txt](https://github.com/waveshare/WM8960-Audio-HAT/files/5286728/log.txt)

Tried above suggestions but cannot git it to install.
Any ideas?

Daveolw, 2020-09-26:
Did a complete reinstall of Raspbian GNU/Linux 9 (stretch), WM8960 install, even more errors, How to solve this?
File of setup and log of errors.
[newinstall.txt](https://github.com/waveshare/WM8960-Audio-HAT/files/5287085/newinstall.txt)