waveshareteam/WM8960-Audio-HAT #7: WM9860 sound card not detected

antonbek89, 2019-11-08:
how i resolve it ?
```
pi@retropie:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@retropie:~ $ uname -a
Linux retropie 4.19.75-v8+ #1270 SMP PREEMPT Tue Sep 24 18:59:17 BST 2019 aarch64 GNU/Linux

```

waveshare, 2019-11-16:
After the installation is complete, you need to reboot

RatherSlowTurtle, 2020-02-05:
I'm getting the same issue as above. Even after reboot aplay -l isn't showing the device.

```
dkms is already the newest version (2.3-2).
git is already the newest version (1:2.11.0-3+deb9u5).
i2c-tools is already the newest version (3.1.2-3).
libasound2-plugins is already the newest version (1.1.1-1).
0 upgraded, 0 newly installed, 0 to remove and 211 not upgraded.

Creating symlink /var/lib/dkms/wm8960-soundcard/1.0/source ->
                 /usr/src/wm8960-soundcard-1.0

DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.14.79+ cannot be found at
/lib/modules/4.14.79+/build or /lib/modules/4.14.79+/source.
Created symlink /etc/systemd/system/sysinit.target.wants/wm8960-soundcard.service → /lib/systemd/system/wm8960-soundcard.service.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
```

RatherSlowTurtle, 2020-02-05:
```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l GNU/Linux

pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- 1a -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

```

miguelwon, 2020-03-03:
I'm having the same error as @RatherSlowTurtle. Any update on this issue?

jiangytcn, 2020-03-23:
exactly same issue with kernel `4.19.109-v7+` on raspberry 3b+

RatherSlowTurtle, 2020-03-23:
> exactly same issue with kernel `4.19.109-v7+` on raspberry 3b+

I was getting an error further up the install script about the kernel version not being found. I got around it by linking another version to the one the install script was looking for.

`sudo ln -s -T /lib/modules/4.19.66+/ /lib/modules/4.14.98+`

jiangytcn, 2020-03-23:
I downloaded the kernel and its header from  https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware 

now the installation is complemented. but still not detected

```pi@pi:~ $ aplay -l | grep wm8960
pi@pi:~ $ lsmod  | grep wm8960
snd_soc_wm8960_soundcard    16384  0
snd_soc_simple_card_utils    16384  1 snd_soc_wm8960_soundcard
snd_soc_wm8960         40960  0
snd_soc_core          192512  4 snd_soc_wm8960_soundcard,snd_soc_simple_card_utils,snd_soc_bcm2835_i2s,snd_soc_wm8960
snd_pcm               102400  5 snd_pcm_dmaengine,snd_soc_bcm2835_i2s,snd_bcm2835,snd_soc_core,snd_soc_wm8960
snd                    73728  10 snd_compress,snd_timer,snd_bcm2835,snd_soc_core,snd_soc_wm8960,snd_pcm
pi@pi:~ $ uname  -r
4.19.97-v7+

```

jiangytcn, 2020-03-24:
finally get it running successfully with latest raspbian os
```
pi@raspberrypi:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~ $ cat /etc/rpi-issue 
Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage2
pi@raspberrypi:~ $ uname -r
4.19.97-v7+```

kld007, 2020-04-02:
Ok nice and how do you solve it?

jiangytcn, 2020-04-03:
old-school way, installed with latest raspbian image

AsebWebDev, 2020-08-04:
Having the same issue with Kernel 5.4.51+

pi@raspberrypi:~/WM8960-Audio-HAT $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~/WM8960-Audio-HAT $ cat /etc/rpi-issue
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2
pi@raspberrypi:~/WM8960-Audio-HAT $ uname -r
5.4.51+

stevewiig, 2023-05-03:
Same issue when trying to install the WM8960 sound card on a Rasberry Pi 4.  Am using the latest image of Raspbian.
"Your kernel headers for kernel 6.1.21-v8+ cannot be found" when installing Waveshare WM8960 sound card

sleiman, 2023-05-16:
@stevewiig any luck? getting the same issue

sleiman, 2023-05-16:
I fixed my problem by doing the following:

Edit config.txt
`cd /boot
sudo nano config.txt
`

Add this line
`arm_64bit=0`

Reboot
`sudo reboot`

Mohit-Ak, 2023-11-07:
> Same issue when trying to install the WM8960 sound card on a Rasberry Pi 4. Am using the latest image of Raspbian. "Your kernel headers for kernel 6.1.21-v8+ cannot be found" when installing Waveshare WM8960 sound card

I had the same issue using raspberry pi os lite 64 bit. Solved using this repo which supportes the 6+ kernel and a specific branch as recommended by their support team:

git clone https://github.com/HinTak/seeed-voicecard -b v6.1