waveshareteam/WM8960-Audio-HAT #9: Headphone jack detection does not work

michnixweiss, 2020-04-22:
I'm using the WM8960 HAT on a Raspi zero using raspbian buster. I've installed the driver like described in the wiki. Everything is working (sound on speaker, on headphone, adjustable with alsa).
BUT when the headphone jack is plugged, it doesn't mute the speaker.
In the datasheet on page 45 the jack detection is mentioned, but I don't have enough knowledge to unterstand how that has to be done. Is it a hardware switch or a software switch? If hardware I think mine is broken, if software does that have to be implemented in the driver, in alsa or done by a script which listens to a gpio?
Thanks a lot

StonedEdge, 2021-01-19:
Did you ever solve this issue? I would also like to figure out how to disable speakers when HP is plugged in.

michnixweiss, 2021-01-21:
Only a workaround with a liste soldering and a script. I can Post details in you want.

Bastler80, 2021-06-03:
I really would like to have the script. is it possible for you to post it?

michnixweiss, 2021-07-02:
Sorry for late reply.
I created a JackDetect.py with following content:
```
#!/usr/bin/python

import RPi.GPIO as GPIO

from signal import pause

import subprocess

 

 

# GPIO initialisieren

GPIO.setmode(GPIO.BCM)

GPIO.setup(26, GPIO.IN)

 

# internen Pullup-Widerstand aktivieren.

GPIO.setup(26, GPIO.IN, pull_up_down = GPIO.PUD_UP)

 

# Callback-Funktion fuer beide Flanken

def measure(channel):

if GPIO.input(26) == 0: # fallende Flanke, Startzeit speichern

bashCMD = "sudo amixer -c 0 sset Speaker,0 0"

output = subprocess.check_output(['bash','-c', bashCMD])

else: # steigende Flanke, Endezeit speichern

bashCMD = "sudo amixer -c 0 sset Speaker,0 122"

output = subprocess.check_output(['bash','-c', bashCMD])

 

# Interrupt fuer beide Flanken aktivieren

GPIO.add_event_detect(26, GPIO.BOTH, callback=measure, bouncetime=300)

 

pause()

```
This should be started as daemon after boot. I did it with a JackDetect.service
```
[Unit]

Description=Headphone Jack Detection Service

After=network.target iptables.service firewalld.service

 

[Service]

User=pi

Group=pi

Restart=always

WorkingDirectory=/home/pi/RPi-Jukebox-RFID

ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/JackDetect.py

 

[Install]

WantedBy=multi-user.target

 
```