waveshareteam/WM8960-Audio-HAT #10: Build Error on kernal 5.4.51-v7+

TerrenceHenry, 2020-08-01:
I'm trying to get my Audio-HAT working with retropie "retropie-buster-4.6-rpi2_rpi3.img".

During install, the sound card code has a few build errors.

make.log file attached.
[make.log](https://github.com/waveshare/WM8960-Audio-HAT/files/5009937/make.log)

iTommix, 2020-08-04:
> I'm trying to get my Audio-HAT working with retropie "retropie-buster-4.6-rpi2_rpi3.img".
> 
> During install, the sound card code has a few build errors.
> 
> make.log file attached.
> [make.log](https://github.com/waveshare/WM8960-Audio-HAT/files/5009937/make.log)

Facing the same issue. Fixed this by changing the the line

`apt-get -y install raspberrypi-kernel-headers raspberrypi-kernel`

to

`apt-get -y install raspberrypi-kernel-headers`

in **install.sh**. This will install the headers for the Kernel 4.xx on my RaspiOS _Buster_
 
Do it on a fresh system without doing 

`apt-get update`

AsebWebDev, 2020-08-04:
Facing the same issue here, any fix for it on current kernel? :-/

NextUrigeller, 2020-08-05:
I have this issues as well with 5.4.51-v7.

iTommix proposal didn't work for me. I downgraded my Pi to 4.19.xx but I still get the kernel-error's during installation.

Is a fix of this planned? To you have a workaround in the meantime?

TerrenceHenry, 2020-08-05:
As, this was my first retropie build, I used the pre-configured as mentioned on the wiki. 
[Wiki](https://www.waveshare.com/wiki/Arcade-C-1P)
[image](https://drive.google.com/open?id=1jOPzfbwsi4rB_dtxIEe1xl4Q-GfquHbb)

Pegasus272, 2020-08-05:
> I have this issues as well with 5.4.51-v7.
> 
> iTommix proposal didn't work for me. I downgraded my Pi to 4.19.xx but I still get the kernel-error's during installation.
> 
> Is a fix of this planned? To you have a workaround in the meantime?

Hey, for me iTommix proposal did work with a raspbian jessie image.

iTommix, 2020-08-05:
> I have this issues as well with 5.4.51-v7.
> 
> iTommix proposal didn't work for me. I downgraded my Pi to 4.19.xx but I still get the kernel-error's during installation.
> 
> Is a fix of this planned? To you have a workaround in the meantime?

In my case i have RaspiOS Buster (2020-05-27-raspios-buster) and my kernel is

```
pi@pyCAR:~ $ uname -a
Linux pyCAR 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux
```
Without updating apt i'll get these kernel-headers to build the wm8960 drivers

AsebWebDev, 2020-08-06:
I downgraded my kernel with

`sudo rpi-update e696d287001a06bea42b11992671f6c337e1385b
`
to Kernel 4.14.20, so the next Kernel i could find to the one shown on the waveshare wm8960 manual:

[Manual](https://www.waveshare.com/w/upload/5/54/WM8960_Audio_HAT_User_Manual_EN.pdf)

Downgrade funktioniert:

```
uname -a
Linux raspberrypi 4.14.20+ #1094 Mon Feb 19 14:43:02 GMT 2018 armv6l GNU/Linux
```

But i get this error:

`Error! There are no instances of module: wm8960-soundcard`

Probably regarded to [Issue 7](https://github.com/waveshare/WM8960-Audio-HAT/issues/7)

Hopefully this gets updated to work with current kernel. :crossed_fingers:

Pegasus272, 2020-08-07:
> In my case i have RaspiOS Buster (2020-05-27-raspios-buster) and my kernel is
>
> ```
> pi@pyCAR:~ $ uname -a
> Linux pyCAR 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux
> ```
>
> Without [...]

You are right, it worked for me also with a fresh 2020-05-27-raspios-buster image.
Just take out the **apt-update** in the installation skript.

And change:

apt-get -y install raspberrypi-kernel-headers raspberrypi-kernel
to
apt-get -y install raspberrypi-kernel-headers

alchek, 2020-08-08:
Is anyone from Waveshare seeing this and can they advise if a fix is being prepared? I saw a number of systems suffer failed audio when the kernel was upgraded and then found the WM8960 drivers wouldn't build.

jchryssanthacopoulos, 2020-08-09:
I'm also getting the same error as @TerrenceHenry (`error: redefinition of 'struct simple_dai_props'`, etc.). I'm running Buster, kernel 5.4.51-v71+. I removed `raspberrypi-kernel` from the install, and it still fails. Any ideas?

NextUrigeller, 2020-08-11:
> You are right, it worked for me also with a fresh 2020-05-27-raspios-buster image.
> Just take out the **apt-update** in the installation skript.
>
> And change:
>
> apt-get -y install [...]

Basicaly this worked. BUT: The soundcard ist usually definitly not the only software and hardware running on the Pi. It is very uncomfortable to not be able to use apt-get upgrade (as the soundcard whould immediatly stop working) for the overall project. For me as Noob it is nearly impossible. Any piece of additional software is a paine as it is not working properly with the older kernel...

It ist disappointing that I bought hardware, with an installation-script not working and no support for actual firmware. Although they make a apt-get update at the beginning. Furthermore no reply on an two weeks old issue from the supplyer.

alchek, 2020-08-11:
> 
> 
> I'm also getting the same error as @TerrenceHenry (`error: redefinition of 'struct simple_dai_props'`, etc.). I'm running Buster, kernel 5.4.51-v71+. I removed `raspberrypi-kernel` from the install, and it still fails. Any ideas?

> 
> 
> I'm also getting the same error as @TerrenceHenry (`error: redefinition of 'struct simple_dai_props'`, etc.). I'm running Buster, kernel 5.4.51-v71+. I removed `raspberrypi-kernel` from the install, and it still fails. Any ideas?

Your problem is that once you're on the later kernel, the install can't work. You need to stop the upgrade to the kernel before it happens. The kernel that comes with the 2020-05-27 buster image will work with the WM8960 drivers so long as you manually change the install.sh script to prevent it upgrading the kernel as other have described.

alchek, 2020-08-11:
> It ist disappointing that I bought hardware, with an installation-script not working and no support for actual firmware. Although they make a apt-get update at the beginning. Furthermore no reply on an two weeks old issue from the supplyer.

Yes - I have five of them, two of them already installed on fairly important monitoring systems where the WM8960 is used for audible alerts. I've had to roll back to a previous backup image of mine from June and I'm now stuck on the 4.19.118-v7+ kernel.

As I bought mine through Amazon.co.uk I've left a review there stating the current situation. I also emailed service@waveshare.com about three days ago but so far without any success.

Brummbude, 2020-08-14:
Hi guys,

I got the same issue. This is the answer I got from service@waveshare.com:

```
Hello,

Please first update the OS by the following command:

sudo apt-get update
And then install the driver again with sudo 
We tested the driver and it should work, sometimes it may require to install it several times (mostly two times) to make it valid.

--
Best regards

Here is Waveshare Service Team, please feel free to contact us with any questions and reviews. 
For fast support, we recommend you to provide the type of product you ask for with your question and please provide your purchase information.

Website: https://www.waveshare.com
Wiki: https://www.waveshare.com/wiki   

Waveshare Service Team
```
So did anyone solved the problem in real life?

Regards

Brummbude, 2020-08-14:
THEY FIX IT. just update the sources from GITHUB. 

@TerrenceHenry I think you can close the issue now

THANKS 2 @waveshare 
BUILD on current Kernel 5.4.51-v7+: 
```
DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make -j4 KERNELRELEASE=5.4.51-v7+ -C /lib/modules/5.4.51-v7+/build M=/var/lib/dkms/wm8960-soundcard/1.0/build.......
cleaning build area....

DKMS: build completed.

snd-soc-wm8960.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.51-v7+/kernel/sound/soc/codecs/

snd-soc-wm8960-soundcard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.51-v7+/kernel/sound/soc/bcm/

depmod....

DKMS: install completed.
mkdir: cannot create directory ‘/etc/wm8960-soundcard’: File exists
ELEASE=5.4.51-v7+Job for wm8960-soundcard.service failed because the control process exited with error code.
See "systemctl status wm8960-soundcard.service" and "journalctl -xe" for details.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------

```

alchek, 2020-08-15:
Yes, they've updated the source files on GitHub.

Now, for me, the installer completed on a single pass on a system based on 2020-05-27-raspios-buster-lite-armfh.img and fully updated (5.4.51-v7+ kernel) before running their install script.

On one of the systems that I'd held back to the 4.19.118-v7+ kernel and had a working WM8960, I did have to run the install script twice for it to complete successfully.

TerrenceHenry, 2020-08-15:
Thank you all, I don't have time atm to test it myself, but based on the above, the issue appears to have been resolved, so I'll close this issue.

closed 2020-08-15 by TerrenceHenry