waveshareteam/WM8960-Audio-HAT #15: Does not work with Raspberry Pi 4b with current Kernel 5.4.72-v7l+ #1356 SMP

bela-a-am, 2020-11-13:
pi@raspberrypi:/var/lib/dpkg/WM8960-Audio-HAT $ sudo ./install.sh
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
raspberrypi-kernel is already the newest version (1.20201022-1).
raspberrypi-kernel-headers is already the newest version (1.20201022-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
dkms is already the newest version (2.6.1-4).
git is already the newest version (1:2.20.1-2+deb10u3).
i2c-tools is already the newest version (4.1-1).
libasound2-plugins is already the newest version (1.1.8-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Creating symlink /var/lib/dkms/wm8960-soundcard/1.0/source ->
                 /usr/src/wm8960-soundcard-1.0

DKMS: add completed.
**Error! echo
Your kernel headers for kernel 5.4.51-v7l+ cannot be found at
/lib/modules/5.4.51-v7l+/build or /lib/modules/5.4.51-v7l+/source.**
Created symlink /etc/systemd/system/sysinit.target.wants/wm8960-soundcard.service → /lib/systemd/system/wm8960-soundcard.service.
**Job for wm8960-soundcard.service failed because the control process exited with error code.
See "systemctl status wm8960-soundcard.service" and "journalctl -xe" for details.**
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------

bela-a-am, 2020-11-13:
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux

xevilstar, 2020-11-27:
https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2020-08-24/

doesn't work either ... same kernel.

have you found any soutions ?

qkum, 2022-03-29:
From my test same with raspberry pi 400 (keyboard raspberry pi 4 version) on Raspberry Pi OS, Q1 2022.