waveshareteam/WM8960-Audio-HAT #18: is this project dead ?

xevilstar, 2020-11-27:
you got two issues about the recent kernel on two latest versions of stable raspberry pi os linux (x86 and x86_64) and you are basically ignoring us .... is this project dead and buried ?

vgromov, 2020-12-30:
Yep, on current RPIOS RPI3b it basically dead: Driver is built OK, but service refises to start:
```
uname -r
5.4.79-v7+

OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2c@7e804000/status 
OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2s@7e203000/status
Dec 30 06:44:31 raspberrypi systemd[1]: wm8960-soundcard.service: Main process exited, code=exited, status=1/FAILURE
```

shawaj, 2021-01-02:
Try the seedstudio repo - https://github.com/respeaker/seeed-voicecard/

As it appears that this HAT is just a clone of that card looking at the device tree overlay - https://github.com/respeaker/seeed-voicecard/blob/master/seeed-2mic-voicecard-overlay.dts