waveshareteam/WM8960-Audio-HAT #23: New WM8960 Soundcard overlay

shawaj, 2021-01-08:
The wm8960-soundcard overlay has been merged into the RasPi kernel...

https://github.com/raspberrypi/linux/pull/4039

Once it gets into the firmware repo (a few days probably):

https://github.com/raspberrypi/firmware/tree/master/boot/overlays

You should be able to use this board by simply adding `dtoverlay=wm8960-soundcard` to the `/boot/config.txt` file.

shawaj, 2021-01-16:
Now available in the official pi firmware - https://github.com/raspberrypi/firmware/blob/master/boot/overlays/wm8960-soundcard.dtbo

asprinwizard, 2021-01-24:
Hi there. Can you advise how to get this running in Libreelec. I have updated to the latest version but the overlay is not there. Perhaps it is not yet using the latest Kernel? I copied the overlay file and added the line but the device does not show up in the list of audio devices. Would be grateful for any help. Thanks.

TobiasVanDyk, 2021-02-07:
> 
> 
> Now available in the official pi firmware - https://github.com/raspberrypi/firmware/blob/master/boot/overlays/wm8960-soundcard.dtbo

Could you please tell us what to add to config.txt I tried the usual dtoverlay=wm8960-soundcard and enabled i2c and i2s but it did not work for kernel 4.10.8.

shawaj, 2021-02-07:
@TobiasVanDyk you'll likely need to be on kernel 5.4.y or above

TobiasVanDyk, 2021-02-07:
> 
> 
> @TobiasVanDyk you'll likely need to be on kernel 5.4.y or above

Thanks - I edited my response I typed kernel 4 by mistake...

JakubVanek, 2021-02-07:
@shawaj It seems that the RPi Foundation's 5.10 kernel does not include the wm8960 module anymore. The following command does not return anything on a freshly updated Raspberry:
```
pi@oggy-pi4:/lib/modules $ find | grep wm8960
```

AIWintermuteAI, 2021-07-30:
@JakubVanek it's because it wasn't compiled by default. 
Now it is 
https://github.com/raspberrypi/linux/issues/4454
However the codec is likely broken and using it will result in problems with recording sound
https://github.com/raspberrypi/linux/issues/4384