waveshareteam/WM8960-Audio-HAT #24: kernel 5.10.11+ problem

dr-ni, 2021-02-05:
DKMS make.log for wm8960-soundcard-1.0 for kernel 5.10.11+ (armv6l)
Fr 5. Feb 14:50:10 CET 2021
make: Verzeichnis „/usr/src/linux-headers-5.10.11+“ wird betreten
  CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.o
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c: In function ‘wm8960_configure_clocking’:
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:747:15: error: implicit declaration of function ‘snd_soc_component_read32’; did you mean ‘snd_soc_component_read’? [-Werror=implicit-function-declaration]
  u16 iface1 = snd_soc_component_read32(component, WM8960_IFACE1);
               ^~~~~~~~~~~~~~~~~~~~~~~~
               snd_soc_component_read
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c: At top level:
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:1367:3: error: ‘const struct snd_soc_dai_ops’ has no member named ‘digital_mute’
  .digital_mute = wm8960_mute,
   ^~~~~~~~~~~~
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:1367:18: error: initialization of ‘int (*)(struct snd_pcm_substream *, struct snd_soc_dai *)’ from incompatible pointer type ‘int (*)(struct snd_soc_dai *, int)’ [-Werror=incompatible-pointer-types]
  .digital_mute = wm8960_mute,
                  ^~~~~~~~~~~
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:1367:18: note: (near initialization for ‘wm8960_dai_ops.prepare’)
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:279: /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.o] Error 1
make: *** [Makefile:1808: /var/lib/dkms/wm8960-soundcard/1.0/build] Fehler 2
make: Verzeichnis „/usr/src/linux-headers-5.10.11+“ wird verlassen

dr-ni, 2021-02-05:
Workaround in wm8960.c:

Replace all
snd_soc_component_read32
to
snd_soc_component_read

and replace L928
static int wm8960_mute(struct snd_soc_dai *dai, int mute)
to
static int wm8960_mute(struct snd_soc_dai *dai, int mute, int direction)

and replace L1367
.digital_mute = wm8960_mute,
to
,.mute_stream = wm8960_mute,

paspo, 2021-02-06:
works for me 👍

TobiasVanDyk, 2021-02-06:
Yes replacing the wm8960.c with the changes made also fixed the audio for me - check the headphone volume = 0 with alsamixer if still no sound

M-t-t-h, 2021-02-15:
work for me too.
Tks To dr-ni for this fix and tks to project creator!

Twisticles, 2021-03-05:
@dr-ni 
> Workaround in wm8960.c:
>
> Replace all
> snd_soc_component_read32
> to
> snd_soc_component_read
>
> and replace L928
> static int wm8960_mute(struct snd_soc_dai *dai, int mute)
> to
> static int wm8960_mute(struct [...]

Apologies, I'm new to this (only started messing with Pi a couple weeks ago) and I can't find any details on how to do this.  Where should I be looking to do these edits to try and fix my audio?
The mic/recording is fine. The HAT is recognised (Card 2).  Just nothing coming out!

TobiasVanDyk, 2021-03-05:
Those have been corrected on the waveshare github.
Just nothing coming out!:
Try terminal "alsamixer", F6 and select card, then adjust headphone volume up  from 0

dr-ni, 2021-03-05:
my pull request has already been merged by waveshare. So everything should be working well now with this repository.

Twisticles, 2021-03-05:
@TobiasVanDyk @dr-ni 
Thanks for the quick replies!  Sadly still no joy for me from the HAT headphone jack nor the JST port.
 Is there a remove command so I can uninstall and try again?  Any pointers on how to debug/trouble shoot?
 
 `**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 2: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
`
So that is my list of playback devices.
I set to default by doing this:
`sudo nano /usr/share/alsa/alsa.conf 
Open the file and find the statement:
defaults.ctl.card 2 
defaults.pcm.card 2`

That is assuming that Card 2 is correct here.
If it matters, when I launch alsamixer it goes straight to pulseaudio, I have to F6 to get to the wm8960. Not sure if that is significant.

dr-ni, 2021-03-05:
you can try this:
cd WM8960-Audio-HAT
sudo ./uninstall.sh

git clone https://github.com/Sybility/WM8960-Audio-HAT.git
cd WM8960-Audio-HAT
sudo ./install.sh

hope this helps

dr-ni, 2021-03-05:
you can also disable internal audio:
cd /etc/modprobe.d
sudo vi alsa-blacklist.conf
Enter the following line

blacklist snd_bcm2835
Save the file
reboot the pi

Twisticles, 2021-03-05:
@TobiasVanDyk @dr-ni 
Thanks for the quick replies!  Sadly still no joy for me from the HAT headphone jack nor the JST port.
 Is there a remove command so I can uninstall and try again?  Any pointers on how to debug/trouble shoot?
 
 `**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 2: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
`
So that is my list of playback devices.
I set to default by doing this:
`sudo vi /usr/share/alsa/alsa.conf 
Open the file and find the statement:
defaults.ctl.card 0 
defaults.pcm.card 0`

> you can also disable internal audio:
> cd /etc/modprobe.d
> sudo vi alsa-blacklist.conf
> Enter the following line
> 
> blacklist snd_bcm2835
> Save the file
> reboot the pi

Eeek, now there are no soundcards detected at all!  

I did this:
cd WM8960-Audio-HAT
sudo ./uninstall.sh

Then this:
git clone https://github.com/Sybility/WM8960-Audio-HAT.git
cd WM8960-Audio-HAT
sudo ./install.sh

Then this:
sudo vi /usr/share/alsa/alsa.conf 
change to:
defaults.ctl.card 2
defaults.pcm.card 2

And again, mic working and recording fine, turned up headphone in alsamixer on the wm8960, but still no audio out of the headphone or jst.

Any more ideas?

Twisticles, 2021-03-05:
@dr-ni 
> you can also disable internal audio:
> cd /etc/modprobe.d
> sudo vi alsa-blacklist.conf
> Enter the following line
> 
> blacklist snd_bcm2835
> Save the file
> reboot the pi

Yeah that was bad...all sound cards disappeared.  After reverting and another fresh install I can't get the WM8960 back at all :(

dr-ni, 2021-03-05:
then remove alsa-blacklist.conf again.
maybe you have to setup pulseaudio corect

dr-ni, 2021-03-05:
what do aplay -l 
and
arecord -l
show?

Twisticles, 2021-03-05:
Firstly, I know this is for issues and not a help forum, so I really appreciate your time :)

I did remove the alsa-blacklist.conf  and the WM8960 didn't come back.  Both aplay -l and arecord -l show default ins and outs now.  Whereas previously they were both hw2,0

**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3

**** List of CAPTURE Hardware Devices ****
(my edit: null shown)

I've now whipped out a spare Pi 4 and a second WM8960 HAT, and tried a completely clean install on that too...it worked as far as testing audio...the recording worked  (sudo arecord -D hw:2,0 -f S32_LE -r 16000 -c 2 test.wav)  and immediately afterwards I went to play the test.wav (sudo aplay -D hw:2,0 test.wav) it failed with no known PCM device on hw2,0!  

I dunno I must be cursed, maybe I am frying these boards or something.  I need sleep...

dr-ni, 2021-03-05:
is paplay working?

Twisticles, 2021-03-05:
> is paplay working?

Embarrased to say I dont know what that is...

dr-ni, 2021-03-05:
same as aplay but pulseaudio 
also check your setup 
playback and record devices cannot be both hw:2,0

dr-ni, 2021-03-05:
Sorry I'm wrong
my setup no pulseaudio blacklist internal audio 
aplay -l 
**** Liste der Hardware-Geräte (PLAYBACK) ****
Karte 0: wm8960soundcard [wm8960-soundcard], Gerät 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0
arecord -l 
**** Liste der Hardware-Geräte (CAPTURE) ****
Karte 0: wm8960soundcard [wm8960-soundcard], Gerät 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Sub-Geräte: 1/1
  Sub-Gerät #0: subdevice #0`

dr-ni, 2021-03-05:
sudo cat /etc/asound.conf
pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "capture"
}

pcm.playback {
    type plug
    slave.pcm "dmixed"
}

pcm.capture {
    type plug
    slave.pcm "array"
}

pcm.dmixed {
    type dmix
    slave.pcm "hw:wm8960soundcard"
    ipc_key 555555 
}

pcm.array {
    type dsnoop
    slave {
        pcm "hw:wm8960soundcard"
        channels 2
    }
    ipc_key 666666
}

janiluuk, 2021-04-19:
i have no sound either. 

decided to change supplier for our products since this is yet another example of waveshare support. as long as it used to work once at some point of time, they consider it as done.