waveshareteam/WM8960-Audio-HAT #27: Error! Bad return status for module build on kernel: 5.10.11+ (armv6l)

agentsimon, 2021-02-10:
Hello
I am trying to install from the card on a Pi Zero running Buster but I get this error
Building module:
cleaning build area.....
make -j1 KERNELRELEASE=5.10.11+ -C /lib/modules/5.10.11+/build M=/var/lib/dkms/wm8960-soundcard/1.0/build.........(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.11+ (armv6l)
Consult /var/lib/dkms/wm8960-soundcard/1.0/build/make.log for more information.
Created symlink /etc/systemd/system/sysinit.target.wants/wm8960-soundcard.service → /lib/systemd/system/wm8960-soundcard.service.
Job for wm8960-soundcard.service failed because the control process exited with error code.
See "systemctl status wm8960-soundcard.service" and "journalctl -xe" for details.
Is there a fix for this? I

dr-ni, 2021-02-10:
can you post /var/lib/dkms/wm8960-soundcard/1.0/build/make.

agentsimon, 2021-02-10:
DKMS make.log for wm8960-soundcard-1.0 for kernel 5.10.11+ (armv6l)
Wed 10 Feb 2021 05:00:50 PM +07
make: Entering directory '/usr/src/linux-headers-5.10.11+'
  CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.o
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c: In function ‘wm8960_configure_clocking’:
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:747:15: error: implicit declaration of function ‘snd_soc_component_read32’; did you mean ‘snd_soc_component_read’? [-Werror=implicit-function-declaration]
  u16 iface1 = snd_soc_component_read32(component, WM8960_IFACE1);
               ^~~~~~~~~~~~~~~~~~~~~~~~
               snd_soc_component_read
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c: At top level:
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:1367:3: error: ‘const struct snd_soc_dai_ops’ has no member named ‘digital_mute’
  .digital_mute = wm8960_mute,
   ^~~~~~~~~~~~
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:1367:18: error: initialization of ‘int (*)(struct snd_pcm_substream *, struct snd_soc_dai *)’ from incompatible pointer type ‘int (*)(struct snd_soc_dai *, int)’ [-Werror=incompatible-pointer-types]
  .digital_mute = wm8960_mute,
                  ^~~~~~~~~~~
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:1367:18: note: (near initialization for ‘wm8960_dai_ops.prepare’)
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:279: /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.o] Error 1
make: *** [Makefile:1808: /var/lib/dkms/wm8960-soundcard/1.0/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.10.11+'

dr-ni, 2021-02-10:
same as here https://github.com/waveshare/WM8960-Audio-HAT/issues/24

agentsimon, 2021-02-10:
and replace L1367
.digital_mute = wm8960_mute,
to
----->>>>,.mute_stream = wm8960_mute,    the comma here?...is it correct?

TobiasVanDyk, 2021-02-10:
> 
> 
> and replace L1367
> .digital_mute = wm8960_mute,
> to
> ----->>>>,.mute_stream = wm8960_mute, the comma here?...is it correct?

Use https://github.com/Sybility/WM8960-Audio-HAT/blob/master/wm8960.c as a replacement
It has all the corrections made already

agentsimon, 2021-02-10:
Thanks...it's working now

Jakar510, 2021-02-11:
> > and replace L1367
> > .digital_mute = wm8960_mute,
> > to
> > ----->>>>,.mute_stream = wm8960_mute, the comma here?...is it correct?
> 
> Use https://github.com/Sybility/WM8960-Audio-HAT/blob/master/wm8960.c as a replacement
> It has all the corrections made already
> 
> such as .mute_stream = wm8960_mute,

Still having issues.... 

`	pi@SPF1:~/audio $ git clone https://github.com/waveshare/WM8960-Audio-HAT
	Cloning into 'WM8960-Audio-HAT'...
	remote: Enumerating objects: 4, done.
	remote: Counting objects: 100% (4/4), done.
	remote: Compressing objects: 100% (4/4), done.
	remote: Total 55 (delta 0), reused 2 (delta 0), pack-reused 51
	Unpacking objects: 100% (55/55), done.

	pi@SPF1:~/audio $ cd WM8960-Audio-HAT

	pi@SPF1:~/audio/WM8960-Audio-HAT $ sudo ./install.sh
	Hit:1 http://packages.microsoft.com/repos/code stable InRelease
	Hit:2 http://archive.raspberrypi.org/debian buster InRelease                                       
	Hit:3 http://raspbian.raspberrypi.org/raspbian buster InRelease                                    
	Reading package lists... Done
	Building dependency tree       
	Reading state information... Done
	All packages are up to date.
	Reading package lists... Done
	Building dependency tree       
	Reading state information... Done
	raspberrypi-kernel is already the newest version (1.20210201-1).
	raspberrypi-kernel-headers is already the newest version (1.20210201-1).
	0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
	Reading package lists... Done
	Building dependency tree       
	Reading state information... Done
	dkms is already the newest version (2.6.1-4).
	git is already the newest version (1:2.20.1-2+deb10u3).
	i2c-tools is already the newest version (4.1-1).
	libasound2-plugins is already the newest version (1.1.8-1).
	0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

	------------------------------
	Deleting module version: 1.0
	completely from the DKMS tree.
	------------------------------
	Done.

	Creating symlink /var/lib/dkms/wm8960-soundcard/1.0/source ->
					 /usr/src/wm8960-soundcard-1.0

	DKMS: add completed.

	Kernel preparation unnecessary for this kernel.  Skipping...

	Building module:
	cleaning build area...
	make -j4 KERNELRELEASE=5.10.11-v7l+ -C /lib/modules/5.10.11-v7l+/build M=/var/lib/dkms/wm8960-soundcard/1.0/build....(bad exit status: 2)
	Error! Bad return status for module build on kernel: 5.10.11-v7l+ (armv7l)
	Consult /var/lib/dkms/wm8960-soundcard/1.0/build/make.log for more information.
	mkdir: cannot create directory ‘/etc/wm8960-soundcard’: File exists
	Job for wm8960-soundcard.service failed because the control process exited with error code.
	See "systemctl status wm8960-soundcard.service" and "journalctl -xe" for details.
	------------------------------------------------------
	Please reboot your raspberry pi to apply all settings
	Enjoy!
	------------------------------------------------------
[... 58 lines cut ...]
	Deleting module version: 1.0
	completely from the DKMS tree.
	------------------------------
	Done.

	Creating symlink /var/lib/dkms/wm8960-soundcard/1.0/source ->
					 /usr/src/wm8960-soundcard-1.0

	DKMS: add completed.

	Kernel preparation unnecessary for this kernel.  Skipping...

	Building module:
	cleaning build area...
	make -j4 KERNELRELEASE=5.10.11-v7l+ -C /lib/modules/5.10.11-v7l+/build M=/var/lib/dkms/wm8960-soundcard/1.0/build....(bad exit status: 2)
	Error! Bad return status for module build on kernel: 5.10.11-v7l+ (armv7l)
	Consult /var/lib/dkms/wm8960-soundcard/1.0/build/make.log for more information.
	mkdir: cannot create directory ‘/etc/wm8960-soundcard’: File exists
	Job for wm8960-soundcard.service failed because the control process exited with error code.
	See "systemctl status wm8960-soundcard.service" and "journalctl -xe" for details.
	------------------------------------------------------
	Please reboot your raspberry pi to apply all settings
	Enjoy!
	------------------------------------------------------
`

Jakar510, 2021-02-11:
Also tried this 

> Workaround in wm8960.c:
> 
> Replace all
> snd_soc_component_read32
> to
> snd_soc_component_read
> 
> and replace L928
> static int wm8960_mute(struct snd_soc_dai *dai, int mute)
> to
> static int wm8960_mute(struct snd_soc_dai *dai, int mute, int direction)
> 
> and replace L1367
> .digital_mute = wm8960_mute,
> to
> ,.mute_stream = wm8960_mute,

Jakar510, 2021-02-11:
reverted back to original 

	pi@SPF1:~/audio/WM8960-Audio-HAT $ cat /var/lib/dkms/wm8960-soundcard/1.0/build/make.log
	DKMS make.log for wm8960-soundcard-1.0 for kernel 5.10.11-v7l+ (armv7l)
	Wed 10 Feb 2021 07:17:49 PM CST
	make: Entering directory '/usr/src/linux-headers-5.10.11-v7l+'
	  CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.o
	  CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.o
	/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c: In function ‘wm8960_configure_clocking’:
	/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:747:15: error: implicit declaration of function ‘snd_soc_component_read32’; did you mean ‘snd_soc_component_read’? [-Werror=implicit-function-declaration]
	  u16 iface1 = snd_soc_component_read32(component, WM8960_IFACE1);
				   ^~~~~~~~~~~~~~~~~~~~~~~~
				   snd_soc_component_read
	/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c: At top level:
	/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:1367:3: error: ‘const struct snd_soc_dai_ops’ has no member named ‘digital_mute’
	  .digital_mute = wm8960_mute,
	   ^~~~~~~~~~~~
	/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:1367:18: error: initialization of ‘int (*)(struct snd_pcm_substream *, struct snd_soc_dai *)’ from incompatible pointer type ‘int (*)(struct snd_soc_dai *, int)’ [-Werror=incompatible-pointer-types]
	  .digital_mute = wm8960_mute,
					  ^~~~~~~~~~~
	/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:1367:18: note: (near initialization for ‘wm8960_dai_ops.prepare’)
	cc1: some warnings being treated as errors
	make[1]: *** [scripts/Makefile.build:279: /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.o] Error 1
	make[1]: *** Waiting for unfinished jobs....
	make: *** [Makefile:1808: /var/lib/dkms/wm8960-soundcard/1.0/build] Error 2
	make: Leaving directory '/usr/src/linux-headers-5.10.11-v7l+'

dr-ni, 2021-02-11:
you can try this:
sudo ./uninstall.sh 

git clone https://github.com/Sybility/WM8960-Audio-HAT.git
cd WM8960-Audio-HAT
sudo ./install.sh

hope this helps

Jakar510, 2021-02-11:
> you can try this:
> sudo ./uninstall.sh
> 
> git clone https://github.com/Sybility/WM8960-Audio-HAT.git
> cd WM8960-Audio-HAT
> sudo ./install.sh
> 
> hope this helps

It works. Thank you!

A-KL, 2021-02-15:
Hey, guys, thanks for the advices. I was able install and use overlay on my Raspberry Pi 3 Model B+!

However, the sound device just disappears after power off\on. Reboot works just fine, but not shutdown.
```
pi@rpi-rover:~/projects/WM8960-Audio-HAT $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 2: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
I think it's somehow related to the fact that the service was never able to start:

```
pi@rpi-rover:~/projects/WM8960-Audio-HAT $ pi@rpi-rover:~/projects/WM8960-Audio-HAT $ systemctl status wm8960-soundcard.service
● wm8960-soundcard.service - WM8960 soundcard service
   Loaded: loaded (/lib/systemd/system/wm8960-soundcard.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-02-15 22:10:47 GMT; 4min 10s ago
  Process: 469 ExecStart=/usr/bin/wm8960-soundcard (code=exited, status=1/FAILURE)
 Main PID: 469 (code=exited, status=1/FAILURE)

Feb 15 22:10:44 rpi-rover systemd[1]: Starting WM8960 soundcard service...
Feb 15 22:10:44 rpi-rover wm8960-soundcard[469]: + exec
Feb 15 22:10:44 rpi-rover wm8960-soundcard[469]: ++ basename /usr/bin/wm8960-soundcard
Feb 15 22:10:47 rpi-rover systemd[1]: wm8960-soundcard.service: Main process exited, code=exited, status=1/FAILURE
Feb 15 22:10:47 rpi-rover systemd[1]: wm8960-soundcard.service: Failed with result 'exit-code'.
Feb 15 22:10:47 rpi-rover systemd[1]: Failed to start WM8960 soundcard service.
```

```
Feb 15 22:18:03 rpi-rover sudo[1286]:       pi : TTY=pts/0 ; PWD=/home/pi/projects/WM8960-Audio-HAT ; USER=root ; COMMAND=/bin/systemctl start wm8960-soundcard.service
Feb 15 22:18:03 rpi-rover sudo[1286]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Feb 15 22:18:03 rpi-rover systemd[1]: Starting WM8960 soundcard service...
-- Subject: A start job for unit wm8960-soundcard.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit wm8960-soundcard.service has begun execution.
--
-- The job identifier is 753.
Feb 15 22:18:03 rpi-rover wm8960-soundcard[1289]: + exec
Feb 15 22:18:03 rpi-rover wm8960-soundcard[1289]: ++ basename /usr/bin/wm8960-soundcard
Feb 15 22:18:03 rpi-rover kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2c@7e804000/status
Feb 15 22:18:04 rpi-rover kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i2s@7e203000/status
Feb 15 22:18:05 rpi-rover kernel: wm8960 1-001a: ASoC: error at snd_soc_component_update_bits on wm8960.1-001a: -121
Feb 15 22:18:05 rpi-rover kernel: wm8960 1-001a: ASoC: error at snd_soc_component_update_bits on wm8960.1-001a: -121
Feb 15 22:18:05 rpi-rover systemd[1]: wm8960-soundcard.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit wm8960-soundcard.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
Feb 15 22:18:05 rpi-rover systemd[1]: wm8960-soundcard.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit wm8960-soundcard.service has entered the 'failed' state with result 'exit-code'.
Feb 15 22:18:05 rpi-rover systemd[1]: Failed to start WM8960 soundcard service.
-- Subject: A start job for unit wm8960-soundcard.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
```

I've also tried a clean raspbian image (also running on 5.10.11-v7+) - still same problem. Did anyone get similar issue?

TobiasVanDyk, 2021-02-16:
> 
> 
> Hey, guys, thanks for the advices. I was able install and use overlay on my Raspberry Pi 3 Model B+!
> 
> However, the sound device just disappears after power off\on. Reboot works just fine, but not shutdown. This is how it looks after reboot but before shutdown:
> 
> ```
> pi@rpi-rover:~/projects/WM8960-Audio-HAT $ aplay -l
> **** List of PLAYBACK Hardware Devices ****
> card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
>   Subdevices: 4/4
>   Subdevice #0: subdevice #0
>   Subdevice #1: subdevice #1
>   Subdevice #2: subdevice #2
>   Subdevice #3: subdevice #3
> card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
>   Subdevices: 4/4
>   Subdevice #0: subdevice #0
>   Subdevice #1: subdevice #1
>   Subdevice #2: subdevice #2
>   Subdevice #3: subdevice #3
> card 2: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
>   Subdevices: 1/1
>   Subdevice #0: subdevice #0
> ```
> 
> I think it's somehow related to the fact that the service was never able to start:
> 
> I've also tried a clean raspbian image (also running on 5.10.11-v7+) - still same problem. Did anyone get similar issue?

Hi
I did a clean install on a Pi 3B+ and the Waveshare WM8960 soundhat works fine after shutdown and reboot
I did notice the same install warning with the Pi 4B 

Some of the details on the Pi 3B+
> ```
>DKMS: install completed.
>Created symlink /etc/systemd/system/sysinit.target.wants/wm8960-soundcard.service ? /lib/systemd/system/wm8960-soundcard.service.
>Job for wm8960-soundcard.service failed because the control process exited with error code.
>
[wm8960install.txt](https://github.com/waveshare/WM8960-Audio-HAT/files/5988725/wm8960install.txt)
See "systemctl status wm8960-soundcard.service" and "journalctl -xe" for details.
> ```

`

A-KL, 2021-02-16:
Thanks, @TobiasVanDyk. I made one more test but wihout any other devices connected and it worked just as you said. I belive my problem was with the power supply 👌

Pythonaire, 2023-10-30:
in my case it was a bit tricky. I use the WM8960 on Raspberry Pi Zero HW - a slow cpu. WM8960 service required Alsa store processes etc (After= Alsa-store.service) while booting. It seems to be, the wm8960-soundcard.service starts to fast, or the other required services are slow. That leads into crashing the wm8960 service. Starting the wm8960 service manually again it works. Now, i set "ExecStartPre=/bin/sleep 5" in the wm8960-soundcard.service Randall works fine.