waveshareteam/WM8960-Audio-HAT #28: Error! Bad return status for module build on kernel: 5.10.11-v7l+ (armv7l)

MartinBettin, 2021-02-12:
Hi,
if I try to install the driver, then I get the following error:

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.10.11-v7l+ -C /lib/modules/5.10.11-v7l+/build M=/var/lib/dkms/wm8960-soundcard/1.0/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.11-v7l+ (armv7l)
Consult /var/lib/dkms/wm8960-soundcard/1.0/build/make.log for more information.
mkdir: das Verzeichnis „/etc/wm8960-soundcard“ kann nicht angelegt werden: Die Datei existiert bereits
Job for wm8960-soundcard.service failed because the control process exited with error code.
See "systemctl status wm8960-soundcard.service" and "journalctl -xe" for details.

I already tried the Workaround from an other issue with no luck.
****************************************
Workaround in wm8960.c:

Replace all
snd_soc_component_read32
to
snd_soc_component_read

and replace L928
static int wm8960_mute(struct snd_soc_dai *dai, int mute)
to
static int wm8960_mute(struct snd_soc_dai *dai, int mute, int direction)

and replace L1367
.digital_mute = wm8960_mute,
to
,.mute_stream = wm8960_mute,
***********************************
This is from the Logfile (make.log)
DKMS make.log for wm8960-soundcard-1.0 for kernel 5.10.11-v7l+ (armv7l)
Fr 12. Feb 14:55:44 CET 2021
make: Verzeichnis „/usr/src/linux-headers-5.10.11-v7l+“ wird betreten
  CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.o
  CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.o
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:1367:3: error: ‘const struct snd_soc_dai_ops’ has no me$
  .digital_stream = wm8960_mute,
   ^~~~~~~~~~~~~~
   mute_stream
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:1367:20: error: initialization of ‘int (*)(struct snd_p$
  .digital_stream = wm8960_mute,
                    ^~~~~~~~~~~
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:1367:20: note: (near initialization for ‘wm8960_dai_ops$
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:279: /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.o] Fehler 1
make[1]: *** Es wird auf noch nicht beendete Prozesse gewartet....
make: *** [Makefile:1808: /var/lib/dkms/wm8960-soundcard/1.0/build] Fehler 2
make: Verzeichnis „/usr/src/linux-headers-5.10.11-v7l+“ wird verlassen

TobiasVanDyk, 2021-02-12:
Do as suggested previously https://github.com/waveshare/WM8960-Audio-HAT/issues/27
 
cd WM8960-Audio-HAT
sudo ./uninstall.sh
sudo reboot

delete any WM8960-AudioHAT folders then

git clone https://github.com/Sybility/WM8960-Audio-HAT.git
cd WM8960-Audio-HAT
sudo ./install.sh

MartinBettin, 2021-02-13:
Wow, thank you! It´s working now!

closed 2021-02-13 by MartinBettin