waveshareteam/WM8960-Audio-HAT #29: Memory Leak warning Prohibiting Service Start

Whogives23, 2021-03-09:
Hi

Running Install.sh produces the following error:
```snd-soc-wm8960.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.17-v7l+/kernel/sound/soc/codecs/

snd-soc-wm8960-soundcard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.17-v7l+/kernel/sound/soc/bcm/

depmod...

DKMS: install completed.
Created symlink /etc/systemd/system/sysinit.target.wants/wm8960-soundcard.service → /lib/systemd/system/wm8960-soundcard.service.
Job for wm8960-soundcard.service failed because the control process exited with error code.
See "systemctl status wm8960-soundcard.service" and "journalctl -xe" for details.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
```

Running  "journalctl -xe"  returns the following:
```
-- The job identifier is 567.
Mar 09 09:09:32 raspberrypi wm8960-soundcard[1890]: + exec
Mar 09 09:09:32 raspberrypi wm8960-soundcard[1890]: ++ basename /usr/bin/wm8960-soundcard
Mar 09 09:09:32 raspberrypi kernel: OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/i
Mar 09 09:09:34 raspberrypi systemd[1]: wm8960-soundcard.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit wm8960-soundcard.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 1.
Mar 09 09:09:34 raspberrypi systemd[1]: wm8960-soundcard.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit wm8960-soundcard.service has entered the 'failed' state with result 'exit-code'.
Mar 09 09:09:34 raspberrypi systemd[1]: Failed to start WM8960 soundcard service.
-- Subject: A start job for unit wm8960-soundcard.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit wm8960-soundcard.service has finished with a failure.
--
-- The job identifier is 567 and the job result is failed.
Mar 09 09:09:34 raspberrypi sudo[606]: pam_unix(sudo:session): session closed for user root
```

I am unable to get the soundcard to be listed by Alsa due to this.

iw2nke, 2021-04-13:
Exactly same issue on Pi 3B+, kernel 5.10.17-v7+.
How to solve it?

sebastientaggart, 2022-10-25:
Same problem here as well.

Pi Zero v1.3
kernel 5.15.61+

sudo dkms status
wm8960-soundcard, 1.0, 5.15.61+, armv61: installed (original_module exists)

In my case this is a completely fresh install with nothing else.

tuxfamily, 2023-01-17:
Same with kernel 5.15.84+

It seems related to a kernel change, discussed here:

https://lore.kernel.org/lkml/VI1PR0701MB68298749C8133A7D69CFDBBFC9F09@VI1PR0701MB6829.eurprd07.prod.outlook.com/T/

But the patch was rejected as, for the maintener, it's an error and not a warning (despite or the "WARNING" in the text).

After investigation, I found that this error/check (`if (!of_node_check_flag(target->np, OF_OVERLAY))`) was added in kernel 5.4.9

https://elixir.bootlin.com/linux/v5.4.8/source/drivers/of/overlay.c
https://elixir.bootlin.com/linux/v5.4.9/source/drivers/of/overlay.c

So this driver should work for all kernels <= 5.4.8.

I will try to donwgrade my kernel... will see.

tuxfamily, 2023-01-17:
Ok, downgrading was too long/complicated, so I started from a fresh install (with `2021-10-30-raspios-bullseye-armhf-lite.img`).

After the first start, I did nothing more than an `apt install git` in order to clone the repo.
Next,  the `install.sh` script, upgraded the kernel from 5.10.x to 5.15.84-v7+ and the built went well.

Unfortunatly, after reboot, same "memory leak" warning, 
but this time, the card was listed with `aplay -l` ... along with the Rpi headphone and hdmi audio.

And, this is apparently the issue: this driver does not play well with others cards/modules.

Adding this:

``ìni
dtparam=audio=off
dtoverlay=vc4-kms-v3d,audio=off
```

to `/boot/config.txt` solved the issue for me (after a reboot of course).

So I guess the real issue is that the install script does not add these params to the `config.txt`.