waveshareteam/WM8960-Audio-HAT #44: Build error on kernel: 5.15.30-v7+ (armv7l)

theNorNi, 2022-04-01:
New kernel version for Raspberry OS Bullseye
raspberrypi-kernel (1:1.20220328-1).
raspberrypi-kernel-headers (1:1.20220328-1).

Build fails with error message:
"Error! Bad return status for module build on kernel: 5.15.30-v7+ (armv7l)"
[make.log](https://github.com/waveshare/WM8960-Audio-HAT/files/8400933/make.log)

bnwlf, 2022-04-02:
Same here at  waveshares 13.1 Magic Mirror with Raspberry OS Bullseye.

dr-ni, 2022-04-06:
I've started to fix the problem:
(see also https://github.com/torvalds/linux/blob/master/sound/soc/generic/simple-card.c)

Compilation is now working:
https://github.com/dr-ni/WM8960-Audio-HAT

However, still some remainining problems:
```
pi@pizero2:~/WM8960-Audio-HAT $ systemctl status wm8960-soundcard.service
 wm8960-soundcard.service - WM8960 soundcard service
     Loaded: loaded (/lib/systemd/system/wm8960-soundcard.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2022-04-06 16:30:29 BST; 7min ago
    Process: 4430 ExecStart=/usr/bin/wm8960-soundcard (code=exited, status=1/FAILURE)
   Main PID: 4430 (code=exited, status=1/FAILURE)
        CPU: 370ms

Apr 06 16:30:27 pilot systemd[1]: Starting WM8960 soundcard service...
Apr 06 16:30:27 pilot wm8960-soundcard[4430]: + exec
Apr 06 16:30:27 pilot wm8960-soundcard[4431]: ++ basename /usr/bin/wm8960-soundcard
Apr 06 16:30:29 pilot systemd[1]: wm8960-soundcard.service: Main process exited, code=exited, status=1/FAILURE
Apr 06 16:30:29 pilot systemd[1]: wm8960-soundcard.service: Failed with result 'exit-code'.
Apr 06 16:30:29 pilot systemd[1]: Failed to start WM8960 soundcard service.
```

dr-ni, 2022-04-06:
update:

forgot to update wm8960.c and wm8960.h
diff: https://github.com/dr-ni/WM8960-Audio-HAT/commit/5f3edc13a3ab591ed2af40a6c0b3b286e312cb8e
so now dmesg is looking better:
[dmesg.txt](https://github.com/waveshare/WM8960-Audio-HAT/files/8430040/dmesg.txt)

but still some problems:
```
pi@pizero2:~/WM8960-Audio-HAT $ arecord test.wav 
ALSA lib pcm_dsnoop.c:638:(snd_pcm_dsnoop_open) unable to open slave
arecord: main:830: Fehler beim Öffnen des Gerätes: Datei oder Verzeichnis nicht gefunden
pi@pizero2:~/WM8960-Audio-HAT $ speaker-test -t wav 

speaker-test 1.2.4

Wiedergabe-Gerät ist default
Stream-Parameter sind 48000 Hz, S16_LE, 1 Kanäle
WAV-Datei(en)
ALSA lib pcm_dmix.c:1075:(snd_pcm_dmix_open) unable to open slave
Fehler beim Öffnen des Gerätes: -2, Datei oder Verzeichnis nicht gefunden
```

dr-ni, 2022-04-07:
update: https://github.com/dr-ni/WM8960-Audio-HAT/commit/4ec150d5a0042727143e3b910b99dd56a3a8011b

I have now some noisy sound on raspberrypi zero 2.
Can anybody test this also on other models and report?

https://github.com/dr-ni/WM8960-Audio-HAT

TobiasVanDyk, 2022-04-07:
Yes thank you - tested on Pi4B waveshare Pi WM8960 Hat- git pull from your own fork - scratchy sound with VLC - but it is getting there!

dr-ni, 2022-04-07:
and recording? is it working?

TobiasVanDyk, 2022-04-07:
yes tested with arecord -v test.wav is working

dr-ni, 2022-04-07:
and speaker-test -t wav -c 2

is the recording stereo and scratching?

dr-ni, 2022-04-07:
is it stereo?

TobiasVanDyk, 2022-04-07:
Stereo

dr-ni, 2022-04-07:
can you please test again:

git pull
git checkout master
sudo ./install.sh
sudo reboot

TobiasVanDyk, 2022-04-07:
Plays nicely with the wav files in /usr/share/sounds - no scratches

But 16bit 44.1kHz wav file still scratchy sounds

dr-ni, 2022-04-07:
what does
`aplay -l `
and
`aplay -L`
show?

TobiasVanDyk, 2022-04-07:
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
default
    Playback/recording through the PulseAudio sound server
playback
capture
dmixed
array
hw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct hardware device without any conversions
plughw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Hardware device with all software conversions
sysdefault:CARD=Headphones
    bcm2835 Headphones, bcm2835 Headphones
    Default Audio Device
dmix:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct sample mixing device
usbstream:CARD=Headphones
    bcm2835 Headphones
    USB Stream Output
hw:CARD=wm8960soundcard,DEV=0
    wm8960-soundcard, bcm2835-i2s-wm8960-hifi wm8960-hifi-0
    Direct hardware device without any conversions
plughw:CARD=wm8960soundcard,DEV=0
    wm8960-soundcard, bcm2835-i2s-wm8960-hifi wm8960-hifi-0
    Hardware device with all software conversions
sysdefault:CARD=wm8960soundcard
    wm8960-soundcard, bcm2835-i2s-wm8960-hifi wm8960-hifi-0
    Default Audio Device
dmix:CARD=wm8960soundcard,DEV=0
    wm8960-soundcard, bcm2835-i2s-wm8960-hifi wm8960-hifi-0
    Direct sample mixing device
usbstream:CARD=wm8960soundcard
    wm8960-soundcard
    USB Stream Output
hw:CARD=vc4hdmi0,DEV=0
    vc4-hdmi-0, MAI PCM i2s-hifi-0
    Direct hardware device without any conversions
plughw:CARD=vc4hdmi0,DEV=0
    vc4-hdmi-0, MAI PCM i2s-hifi-0
    Hardware device with all software conversions
sysdefault:CARD=vc4hdmi0
    vc4-hdmi-0, MAI PCM i2s-hifi-0
    Default Audio Device
hdmi:CARD=vc4hdmi0,DEV=0
    vc4-hdmi-0, MAI PCM i2s-hifi-0
    HDMI Audio Output
dmix:CARD=vc4hdmi0,DEV=0
    vc4-hdmi-0, MAI PCM i2s-hifi-0
    Direct sample mixing device
usbstream:CARD=vc4hdmi0
    vc4-hdmi-0
    USB Stream Output
hw:CARD=vc4hdmi1,DEV=0
    vc4-hdmi-1, MAI PCM i2s-hifi-0
    Direct hardware device without any conversions
plughw:CARD=vc4hdmi1,DEV=0
    vc4-hdmi-1, MAI PCM i2s-hifi-0
    Hardware device with all software conversions
sysdefault:CARD=vc4hdmi1
    vc4-hdmi-1, MAI PCM i2s-hifi-0
    Default Audio Device
hdmi:CARD=vc4hdmi1,DEV=0
    vc4-hdmi-1, MAI PCM i2s-hifi-0
    HDMI Audio Output
dmix:CARD=vc4hdmi1,DEV=0
    vc4-hdmi-1, MAI PCM i2s-hifi-0
    Direct sample mixing device
usbstream:CARD=vc4hdmi1
    vc4-hdmi-1
    USB Stream Output
pi@raspberrypi:~ $

TobiasVanDyk, 2022-04-07:
when using audacious getting much better quality sound - have to adjust the two volume (audacious and main) controls - still some background crackle but much less

bnwlf, 2022-04-07:
Hi folks,
I hadn't the time to check the new fixes. Since my post i found this repo: https://github.com/HinTak/seeed-voicecard. It works at least somehow. (But I will try it in the next days. Thanks for that!)

The wm8960-soundcard.service is there also failing (status 99). Do you know the sense of this service since the driver?

dr-ni, 2022-04-07:
is aplay -D dmixed pcm1644s.wav playing clean?

dr-ni, 2022-04-07:
@bnwlf
sorry I'm not familar with this

TobiasVanDyk, 2022-04-08:
Yes it plays clean no noise - I used another 16bit/44.1kHz file:

pi@raspberrypi:~ $ aplay -D dmixed pcm16LE44100.wav
pcm16LE44100.wav: No such file or directory
pi@raspberrypi:~ $ aplay -D dmixed 1.wav
Playing WAVE '1.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
Warning: rate is not accurate (requested = 44100Hz, got = 48000Hz)
         please, try the plug plugin (-Dplug:dmixed)

I stress again that using VLC with ALSA + wm8960-soundcard in VLC preferences->Audio-settings solve the problem of noisy playback completely

dr-ni, 2022-04-08:
I have actually no clue about that noise is it only with pulseaudio?

you have to play like this:
aplay -D plug:dmixed audio.wav

TobiasVanDyk, 2022-04-08:
I assumed it is the same noise you experienced here/

[dr-ni](https://github.com/dr-ni) commented [23 hours ago](https://github.com/waveshare/WM8960-Audio-HAT/issues/44#issuecomment-1091655301)
update: https://github.com/dr-ni/WM8960-Audio-HAT/commit/4ec150d5a0042727143e3b910b99dd56a3a8011b
I have now some noisy sound on raspberrypi zero 2.
Can anybody test this also on other models and report?
https://github.com/dr-ni/WM8960-Audio-HAT

dr-ni, 2022-04-08:
I have now no noise with dmixed so maybe only a resampling problem.
I don't have pulseaudio and no X

TobiasVanDyk, 2022-04-08:
Yes I am happy - it now plays nicely
I am hoping for others to comment on their experience?

dr-ni, 2022-04-08:
yes, would be interesting how it is working on other systems

https://github.com/dr-ni/WM8960-Audio-HAT

dr-ni, 2022-04-08:
I've now also tested https://github.com/HinTak/seeed-voicecard
It is working very well

@TobiasVanDyk
do you have the noise with this driver?

TobiasVanDyk, 2022-04-09:
I tried to but it does not like my WM8960 waveshare hat
dmesg | grep wm8960 gives
[   17.295029] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm8960-hifi: -22
[   17.295045]  bcm2835-i2s-wm8960-hifi: ASoC: soc_pcm_hw_params() failed (-22)
[   17.295750] wm8960 1-001a: failed to configure clock

I think it either needs a custom kernel and/or a Seed HAT and/or changed install.sh
I attach the install.log
[install3.txt](https://github.com/waveshare/WM8960-Audio-HAT/files/8456371/install3.txt)

dr-ni, 2022-04-09:
can you uninstall your WM8960-Audio-HAT (sudo ./uninstall.sh)
then cd seeed-voicecard
make cltean
make
and send me the output?

TobiasVanDyk, 2022-04-09:
Hi dr-ni
I had a completely new installation - no previous wm8960 drivers installed
The install3.txt above is the result of that clean, new install

TobiasVanDyk, 2022-04-09:
Ok I made a mistake - when I did the git clone from HinTak then the drivers works first time - no need to put it into ALSA
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

dr-ni, 2022-04-09:
ok so my fork is still with errors...

TobiasVanDyk, 2022-04-09:
Yes sorry about the mistake I made....

bnwlf, 2022-04-19:
Hi @dr-ni i had some time to test your driver,.. 

- [ x] compliation without errors
- [ x] recording with arecored is working
- [ ] aplay the on the recorded file causes a scratch noise and causes then an instand reboot.

so I would say its close,.. Thanks for your work.

bnwlf, 2022-04-20:
Success,.. new install helped!

TobiasVanDyk, 2022-05-02:
Using the new 1 May 2022 Waveshare Drivers works perfectly:

git clone https://github.com/waveshare/WM8960-Audio-HAT
cd WM8960-Audio-HAT
sudo ./install.sh
sudo reboot

After the reboot aplay -l should show the wm8960 soundcard and plays pulseaudio and alsa with no distortion

JamesCHub, 2024-07-12:
> Big Thanks @dr-ni for your work. Now i can use the module in the Raspi Desktop although i use the newer dtoverlay=vc4-kms-v3d in the config.txt. And this without noise. I wish to sum up all steps: I uninstalled the unworking WM8960-Audio-HAT from WAVESHARE and deleted the files that i downloaded from the WAVESHARE repo. I made `git clone https://github.com/dr-ni/WM8960-Audio-HAT` in my user directory of my Raspi 4 (32bit Desktop), (attatched with the WM8960 audio module and two speakers and a 7 inch touch display). `cd WM8960-Audio-HAT` `sudo ./install.sh` `sudo reboot`
> 
> Ignore the message "failed to load i2s-mmap" as @TobiasVanDyk mentined. Then, in the grafical desktop of the raspi you make a rigth click on the speaker symbol and select the wm8960-soundcard. After this you can open Chromium and play a youtube video. You will hear the music, but with some noise.
> 
> The next steps will install other drivers for the wm8960-soundcard. Only a new driver, the rest should come from the repo of @dr-ni and stay.
> 
> First i run `cd WM8960-Audio-HAT` and `sudo ./uninstall.sh` to uninstall the driver from @dr-ni . Then `cd ..` and `git clone https://github.com/HinTak/seeed-voicecard` `cd seeed-voicecard` `make clean` `make` `sudo ./install.sh`
> 
> The install.sh script above shows some errors in the midle of working. **You need to reboot and run it again**.
> 
> Then the necessary drivers are installed. Now you can run the install.sh from @dr-ni again, but before i changed this script a little bit. I opened WM8960-Audio-HAT/install.sh and changed this line: `install_module "./" "wm8960-soundcard"` to `#install_module "./" "wm8960-soundcard"`
> 
> Then run again the install script: `cd WM8960-Audio-HAT` `sudo ./install.sh`
> 
> I think a cold boot is necessary. Finish! Now you can open Chromium in Raspi Desktop and hear a youtube video without additional noise. Thanks to above two repos. Thanks @dr-ni .

This worked for me - pi4 with Rasbian 11 (bullseye), 6.1.21-v71+  

Waveshare installer did not work for me. (July 2024)