waveshareteam/WM8960-Audio-HAT #45: Error! Bad return status for module build on kernel: 5.15.32-v7l+ (armv7l)

gab75ayr, 2022-04-06:
Hello,
The WM8960-Audio-HAT board was working fine on Raspberry PI 4B with Raspbian OS bullseye, until 4 days ago when a system software update was installed.
I'm trying to reinstall the board software, but I get this error:

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.15.32-v7l+ -C /lib/modules/5.15.32-v7l+/build M=/var/lib/dkms/wm8960-soundcard/1.0/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.32-v7l+ (armv7l)
Consult /var/lib/dkms/wm8960-soundcard/1.0/build/make.log for more information.

Is there a fix for this?

dr-ni, 2022-04-06:
this is same as issue #44

vvstubbs, 2022-04-07:
There apparently is not real fix for the problem. I gave up returned the module, go with a ReSpeaker module.

dr-ni, 2022-04-07:
it's only a driver problem because of changes in the kernel. There are the same problems with the seeed-voicecard you can run both cards with both drivers...

.
pilot@pilot:~ $ play /var/www/html/audio/au
auflege

gab75ayr, 2022-04-11:
> this is same as issue #44

Thanks for your reply!
I try your fix on my Raspbian OS Bullseye 32 bit and it's work.
I try on Raspbian OS Bullseye 64 bit and dosn't work.
Do you have any ideas to fix the problem on the 64 bit OS?

qkum, 2022-06-06:
i just got the same error a month later

`Static hostname: XXXXX
         Icon name: computer
        Machine ID: XXXXX
           Boot ID: 2ca38ee09d5f488789654a4bbec8f28b
  Operating System: Raspbian GNU/Linux 10 (buster)
            Kernel: Linux 5.10.103-v7+
      Architecture: arm
      32 bit
`

will try the suggested fix now

qkum, 2022-06-06:
> git clone https://github.com/HinTak/seeed-voicecard also works for the 64bit kernel 5.15.xx

it works perfectly 💯