waveshareteam/WM8960-Audio-HAT #52: Build error on kernel: 6.1.21-v7+ (armv7l) (pi2zeroW)

Fysek, 2023-12-10:
I am working on PI 2 zero W with WM8960 Audio HAT https://www.waveshare.com/wiki/WM8960_Audio_HAT
raspberrypi-kernel : (1:1.20230405-1)
raspberrypi-kernel-headers: (1:1.20230405-1)

I did this: 
sudo apt-get dist-upgrade
sudo apt-get update
git clone https://github.com/waveshare/WM8960-Audio-HAT
cd WM8960-Audio-HAT
sudo nano wm8960-soundcard.dts
Change target = <&i2c1>; to <&i2c0> ; line 25
sudo ./install.sh
sudo reboot -> **nothing changed**

- "uname -a": Linux pizero 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
-  systemctl status wm8960-soundcard.service:
```shell
wm8960-soundcard.service - WM8960 soundcard service
     Loaded: loaded (/lib/systemd/system/wm8960-soundcard.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sun 2023-12-10 18:40:20 CET; 12min ago
    Process: 369 ExecStart=/usr/bin/wm8960-soundcard (code=exited, status=99)
   Main PID: 369 (code=exited, status=99)
        CPU: 716ms

Dec 10 18:40:17 pizero systemd[1]: Starting WM8960 soundcard service...
Dec 10 18:40:17 pizero wm8960-soundcard[369]: + exec
Dec 10 18:40:17 pizero wm8960-soundcard[400]: ++ basename /usr/bin/wm8960-soundcard
Dec 10 18:40:20 pizero systemd[1]: wm8960-soundcard.service: Main process exited, code=exited, status=99/n/a
Dec 10 18:40:20 pizero systemd[1]: wm8960-soundcard.service: Failed with result 'exit-code'.
Dec 10 18:40:20 pizero systemd[1]: Failed to start WM8960 soundcard service.
``` 

Logs:
```shell
cat /var/lib/dkms/wm8960-soundcard/1.0/6.1.21-v7+/armv7l/log/make.log
DKMS make.log for wm8960-soundcard-1.0 for kernel 6.1.21-v7+ (armv7l)
Sun 10 Dec 18:39:13 CET 2023
make: Entering directory '/usr/src/linux-headers-6.1.21-v7+'
  CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.o
  CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.o
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c: In function ‘wm8960_hw_params’:
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:882:6: warning: this statement may fall through [-Wimplicit-fallthrough=]
  882 |   if ((iface & 0x3) != 0) {
      |      ^
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:887:2: note: here
  887 |  default:
      |  ^~~~~~~
  LD [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/snd-soc-wm8960-soundcard.o
  LD [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/snd-soc-wm8960.o
  MODPOST /var/lib/dkms/wm8960-soundcard/1.0/build/Module.symvers
  CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/snd-soc-wm8960-soundcard.mod.o
  CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/snd-soc-wm8960.mod.o
  LD [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/snd-soc-wm8960-soundcard.ko
  LD [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/snd-soc-wm8960.ko
make: Leaving directory '/usr/src/linux-headers-6.1.21-v7+'
```

YuMingC233, 2023-12-22:
I encountered the same `wm8960-soundcard.service error code 99` issue, but I just did a test and it doesn't seem to affect any related functionalities, such as running commands like aplay xxx.wav, etc. It's really strange, isn't it?

Fysek, 2023-12-22:
Interesting finding @YuMingC233, did you try to run **alsamixer** to check if raspbian sees it as a sound card? Here's a manual https://www.waveshare.com/wiki/WM8960_Audio_HAT
Btw. what is your hardware and kernel version?

YuMingC233, 2023-12-22:
### Based on the results, it is able to **very accurately recognize it** as a sound card:

[image]

[image]

### My **hardware**:

Raspberry Pi Zero 2

WM8960 Audio HAT

[image]

After assembly

[image]

### My **kernel version**:

Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr 3 17:24:16 BST 2023 aarch64 GNU/Linux

### How I discovered this anomaly

When I normally boot it up, there is an error at the end of the startup, like this:

[image]

Further error information

[image]

Looks identical, doesn't it? 😏😏

When I tried to reboot:

[image]

Ok, it failed.

When I checked `journalctl -xe` to investigate further:

[image]

Memory leak? But when I checked `htop`, I found no applications hogging system resources:

[image]

However, when I tried to reboot it again, the same error occurred.

Actually, this issue is not "fatal." When I use `aplay xxx.wav` in this state, it can play correctly, and even more advanced "real-time listening" commands like `arecord -f cd | sudo arecord -f cd | aplay` work fine.

Note, none of my commands explicitly specify the hardware.

@Fysek  You might also try testing to see if you can replicate it.

Cheers 🥂

Subuday, 2024-03-14:
I have been experience the same problem.

Subuday, 2024-03-14:
Okay, it looks like I figured out what's problem.

Subuday, 2024-03-14:
On my Raspberry Pi Zero 2 W, I have two devices: 
```
card 0: vc4hdmi
card 1: wm8960soundcard
```
You can check this using `aplay -l`. 

Then I checked logs of wm8960-soundcard.service:
alsactl restore
```
No state is present for card vc4hdmi
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Found hardware: "vc4-hdmi" "" "" "" ""
Hardware is initialized using a generic method
No state is present for card vc4hdmi
```
So it looks like alsactl tries to restore by default with card 0 that is vc4hdmi instead wm8960soundcard.

To fix starting of this service, just modify it to launch with the correct card. In my case it is `alsactl restore -c 1`

Subuday, 2024-03-14:
Maybe we need some code that would either disabling vc4hdmi or setting proper card by searching in system wm8960soundcard.

jcoe2412, 2024-10-06:
I have a similar problem, but the fix does not work for me , I am getting an error 

jcoe2412@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: wm8960soundcard [wm8960-soundcard], device 0: 1f000a0000.i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
jcoe2412@raspberrypi:~ $ alsactl restore -c 2
alsactl: state_lock:167: file /var/lib/alsa/asound.state lock error: Permission denied
alsactl: load_state:1689: Cannot open /var/lib/alsa/asound.state for reading: Permission denied
alsactl: card_lock:196: card 2 lock error: Permission denied
jcoe2412@raspberrypi:~ $ sudo alsactl restore -c 2
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:2 use case configuration -2

Any idea how to fix this ? MAny thanks.

jcoe2412, 2024-10-06:
I forgot to mention above that my problem is on the PI5, I guess it is the same though.

jcoe2412, 2024-10-07:
if the issue is caused by the HDMI soundcards, wouldn't the following fix the problem if we add ",noaudio" to /boot/firmware/config.txt 

dtoverlay=vc4-kms-v3d,noaudio

see https://www.raspberrypi.com/documentation/computers/config_txt.html#hdmi-audio

lukegoeree, 2024-11-04:
Just configured my first Audio HAT for the Pi ZERO2W and can confirm this step worked.

To clarify, had the same "symptoms", everything ran fine, including the test WAV audio, and even MP3s via mpg123, without issue. Recognised the soundcard fine, and even defaulted the audio hat to card 0. But also had the vc4hdmi as card 1. systemctl would wm8960-soundcard status as exited, and status=99.

confirmed setting vc4hdmi to no aduio, then reboot resolved issue.

/boot/firmware/config.txt:

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d,noaudio
max_framebuffers=2

status=0/SUCCESS

thanks for the info.