waveshareteam/WM8960-Audio-HAT #54: Kernel build fails on kernel 6.6.20+rpt-rpi-v7 (armv7l)

sa0bux, 2024-03-17:
DKMS make.log for wm8960-soundcard-1.0 for kernel 6.6.20+rpt-rpi-v7 (armv7l)
Sun Mar 17 09:03:42 AM CET 2024
make: Entering directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v7'
  CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.o
  CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.o
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c: In function ‘asoc_simple_parse_dai’:
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:92:15: error: too few arguments to function ‘snd_soc_of_get_dai_name’
   92 |         ret = snd_soc_of_get_dai_name(node, &dlc->dai_name);
      |               ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/src/linux-headers-6.6.20+rpt-common-rpi/include/sound/simple_card.h:12,
                 from /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:15:
/usr/src/linux-headers-6.6.20+rpt-common-rpi/include/sound/soc.h:1335:5: note: declared here
 1335 | int snd_soc_of_get_dai_name(struct device_node *of_node,
      |     ^~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/scripts/Makefile.build:248: /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.o] Erro
r 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:1527:21: error: initialization of ‘int (*)(struct i2c_client *)’ from incompatible pointer type ‘int (
*)(struct i2c_client *, const struct i2c_device_id *)’ [-Werror=incompatible-pointer-types]
 1527 |         .probe =    wm8960_i2c_probe,
      |                     ^~~~~~~~~~~~~~~~
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:1527:21: note: (near initialization for ‘wm8960_i2c_driver.probe’)
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c: In function ‘wm8960_hw_params’:
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:882:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
  882 |                 if ((iface & 0x3) != 0) {
      |                    ^
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.c:887:9: note: here
  887 |         default:
      |         ^~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/scripts/Makefile.build:248: /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:1938: /var/lib/dkms/wm8960-soundcard/1.0/build] Error 2
make: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:246: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v7'

christiscarborough, 2024-03-22:
I fixed this by running ./uninstall.sh in the driver directory, installing the 6.6 kernel, then re-running ./install.sh.  This may help for you.

sa0bux, 2024-03-31:
> I fixed this by running ./uninstall.sh in the driver directory, installing the 6.6 kernel, then re-running ./install.sh. This may help for you.

Doesn't work for me , you can install the latest kernel but it will fail again when you try build the wm8960 kernel module in 6.6.20.
The definition of "snd_soc_of_get_dai_name" has changed in soc.h compared to 6.1 kernel.

6.1:
 int snd_soc_of_get_dai_name(struct device_node *of_node,
                            const char **dai_name);

6.6:

int snd_soc_of_get_dai_name(struct device_node *of_node,
                            const char **dai_name, int index);

sa0bux, 2024-04-01:
I made a quick and dirty fix so that the module could be compiled under 6.6.20+rpt-rpi-v7
Audio works but I'm not a programmer.

`diff --git a/wm8960-soundcard.c b/wm8960-soundcard.c
index e1df3db..84f6230 100755
--- a/wm8960-soundcard.c
+++ b/wm8960-soundcard.c
@@ -89,7 +89,7 @@ static int asoc_simple_parse_dai(struct device_node *node,
         * 2) user need to rebind Sound Card everytime
         *    if he unbinded CPU or Codec.
         */
-       ret = snd_soc_of_get_dai_name(node, &dlc->dai_name);
+       ret = snd_soc_of_get_dai_name(node, &dlc->dai_name, 0);
        if (ret < 0)
                return ret;

diff --git a/wm8960.c b/wm8960.c
index b5bda3b..7c2c13a 100755
--- a/wm8960.c
+++ b/wm8960.c
@@ -1440,8 +1440,8 @@ static void wm8960_set_pdata_from_of(struct i2c_client *i2c,
                pdata->shared_lrclk = true;
 }

-static int wm8960_i2c_probe(struct i2c_client *i2c,
-                           const struct i2c_device_id *id)
+static int wm8960_i2c_probe(struct i2c_client *i2c)
+
 {
        struct wm8960_data *pdata = dev_get_platdata(&i2c->dev);
        struct wm8960_priv *wm8960;
`