waveshareteam/WM8960-Audio-HAT #62: Bookworm w/ Kernel 6.6.47: Build successful, Audio works; Service and Headphones fail

Tilogorn, 2024-09-25:
Hi guys,

I've read through all "kernel xy build fails" issues, but I have the feeling this is a new issue. Building the binary works, using the soundcard as speaker output in the system works. But the phone jack does not work and (re)starting the `wm8960-soundcard.service` results in exit code `99` (which is related to the non-functional phone jack, I believe).

My setup: Raspberry PI 5 with WM9860 HAT and recent Rasperry Pi OS (released 2024-07-24)
```shell
$ uname -a
Linux spotify-player-raspbian 6.6.47+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64 GNU/Linux
```

<details>
  <summary>Output of install.sh</summary>

  ```shell
  OK:1 http://deb.debian.org/debian bookworm InRelease
  OK:2 http://deb.debian.org/debian-security bookworm-security InRelease
  OK:3 http://archive.raspberrypi.com/debian bookworm InRelease
  OK:4 http://deb.debian.org/debian bookworm-updates InRelease
  Paketlisten werden gelesen… Fertig
  Abhängigkeitsbaum wird aufgebaut… Fertig
  Statusinformationen werden eingelesen… Fertig
  Aktualisierung für 1 Paket verfügbar. Führen Sie »apt list --upgradable« aus, um es anzuzeigen.
  Paketlisten werden gelesen… Fertig
  Abhängigkeitsbaum wird aufgebaut… Fertig
  Statusinformationen werden eingelesen… Fertig
  raspberrypi-kernel-headers ist schon die neueste Version (1:1.20230405-1).
  0 aktualisiert, 0 neu installiert, 0 zu entfernen und 1 nicht aktualisiert.
  Paketlisten werden gelesen… Fertig
  Abhängigkeitsbaum wird aufgebaut… Fertig
  Statusinformationen werden eingelesen… Fertig
  dkms ist schon die neueste Version (3.0.10-8+deb12u1).
  git ist schon die neueste Version (1:2.39.5-0+deb12u1).
  i2c-tools ist schon die neueste Version (4.3-2+b3).
  libasound2-plugins ist schon die neueste Version (1.2.7.1-1).
  0 aktualisiert, 0 neu installiert, 0 zu entfernen und 1 nicht aktualisiert.
  Error! The module/version combo: wm8960-soundcard-1.0 is not located in the DKMS tree.
  Creating symlink /var/lib/dkms/wm8960-soundcard/1.0/source -> /usr/src/wm8960-soundcard-1.0
  Sign command: /lib/modules/6.6.47+rpt-rpi-2712/build/scripts/sign-file
  Signing key: /var/lib/dkms/mok.key
  Public certificate (MOK): /var/lib/dkms/mok.pub

  Building module:
  Cleaning build area...
  make -j4 KERNELRELEASE=6.6.47+rpt-rpi-2712 -C /lib/modules/6.6.47+rpt-rpi-2712/build M=/var/lib/dkms/wm8960-soundcard/1.0/build...
  Signing module /var/lib/dkms/wm8960-soundcard/1.0/build/snd-soc-wm8960.ko
  Signing module /var/lib/dkms/wm8960-soundcard/1.0/build/snd-soc-wm8960-soundcard.ko
  Cleaning build area...

  snd-soc-wm8960.ko.xz:
  Running module version sanity check.
   - Original module
   - Installation
     - Installing to /lib/modules/6.6.47+rpt-rpi-2712/updates/dkms/

  snd-soc-wm8960-soundcard.ko.xz:
  Running module version sanity check.
   - Original module
   - Installation
     - Installing to /lib/modules/6.6.47+rpt-rpi-2712/updates/dkms/
  depmod...
  Created symlink /etc/systemd/system/sysinit.target.wants/wm8960-soundcard.service → /lib/systemd/system/wm8960-soundcard.service.
  Job for wm8960-soundcard.service failed because the control process exited with error code.
  See "systemctl status wm8960-soundcard.service" and "journalctl -xeu wm8960-soundcard.service" for details.
  ------------------------------------------------------
  Please reboot your raspberry pi to apply all settings
  Enjoy!
  ------------------------------------------------------
  ```
</details>

**Output of journalctl -xeu wm8960-soundcard.service**

```shell
× wm8960-soundcard.service - WM8960 soundcard service
     Loaded: loaded (/lib/systemd/system/wm8960-soundcard.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Wed 2024-09-25 10:21:10 CEST; 51s ago
    Process: 3262 ExecStart=/usr/bin/wm8960-soundcard (code=exited, status=99)
   Main PID: 3262 (code=exited, status=99)
        CPU: 102ms

Sep 25 10:21:08 spotify-player-raspbian systemd[1]: Starting wm8960-soundcard.service - WM8960 soundcard service...
Sep 25 10:21:08 spotify-player-raspbian wm8960-soundcard[3262]: + exec
Sep 25 10:21:08 spotify-player-raspbian wm8960-soundcard[3263]: ++ basename /usr/bin/wm8960-soundcard
Sep 25 10:21:10 spotify-player-raspbian systemd[1]: wm8960-soundcard.service: Main process exited, code=exited, status=99/n/a
Sep 25 10:21:10 spotify-player-raspbian systemd[1]: wm8960-soundcard.service: Failed with result 'exit-code'.
Sep 25 10:21:10 spotify-player-raspbian systemd[1]: Failed to start wm8960-soundcard.service - WM8960 soundcard service.
```

**Output of systemctl status wm8960-soundcard.service**

```shell
Sep 25 10:21:08 spotify-player-raspbian systemd[1]: Starting wm8960-soundcard.service - WM8960 soundcard service...
░░ Subject: A start job for unit wm8960-soundcard.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit wm8960-soundcard.service has begun execution.
░░ 
░░ The job identifier is 1144.
Sep 25 10:21:08 spotify-player-raspbian wm8960-soundcard[3262]: + exec
Sep 25 10:21:08 spotify-player-raspbian wm8960-soundcard[3263]: ++ basename /usr/bin/wm8960-soundcard
Sep 25 10:21:10 spotify-player-raspbian systemd[1]: wm8960-soundcard.service: Main process exited, code=exited, status=99/n/a
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ An ExecStart= process belonging to unit wm8960-soundcard.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 99.
Sep 25 10:21:10 spotify-player-raspbian systemd[1]: wm8960-soundcard.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit wm8960-soundcard.service has entered the 'failed' state with result 'exit-code'.
Sep 25 10:21:10 spotify-player-raspbian systemd[1]: Failed to start wm8960-soundcard.service - WM8960 soundcard service.
░░ Subject: A start job for unit wm8960-soundcard.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit wm8960-soundcard.service has finished with a failure.
░░ 
░░ The job identifier is 1144 and the job result is failed.

```

I can narrow it down. One can just do this in the cloned repo:

```shell
~/WM8960-Audio-HAT $ sudo ./wm8960-soundcard
+ exec
++ basename ./wm8960-soundcard
~/WM8960-Audio-HAT $ echo $?
99
```

I am a software developer myself, but my knowledge of C is limited. I cannot find the exit code of `99` in `wm8960-soundcard.c`. But I am willing to change/recompile/add debug output if anyone guides me to get a more precise error message.

Since the combination of a small amplifier with speaker and headphone output, powered by the Pi on a single HAT seems a unique combination on the market, I am willing to support the fix of this issue financially. After all, it will improve the Christmas present for my son. :-)

jonathanpiette, 2024-09-25:
I have the same issue, tried with a Raspberry Pi 4, Raspberry Pi Zero W2 with 3 different HAT. I install a Debian lite via Rasberry Pi imager, follow the documentation to install the board and have this error 99

Tilogorn, 2024-09-26:
Can one of the maintainers confirm that the service issue might relate to the phone jack issue?I am just assuming this because the WM8960 wiki page tells to restart the service if the phone jack does not work.

SofaSurfer76, 2024-10-23:
I have same issue, Rpi4 with Raspiaudio HAT (based on WM8960-soundcard). Reinstalled OS and started from scratch many times, still cannot go further. I have notice that in most of the cases aplay and arecord works fine.

```
sudo systemctl status wm8960-soundcard.service
× wm8960-soundcard.service - WM8960 soundcard service
     Loaded: loaded (/lib/systemd/system/wm8960-soundcard.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Thu 2024-10-17 21:46:04 CEST; 23h ago
   Main PID: 676 (code=exited, status=99)
        CPU: 475ms

Oct 17 21:46:01 assistant systemd[1]: Starting wm8960-soundcard.service - WM8960 soundcard service...
Oct 17 21:46:01 assistant wm8960-soundcard[676]: + exec
Oct 17 21:46:01 assistant wm8960-soundcard[679]: ++ basename /usr/bin/wm8960-soundcard
Oct 17 21:46:04 assistant systemd[1]: wm8960-soundcard.service: Main process exited, code=exited, status=99/n/a
Oct 17 21:46:04 assistant systemd[1]: wm8960-soundcard.service: Failed with result 'exit-code'.
Oct 17 21:46:04 assistant systemd[1]: Failed to start wm8960-soundcard.service - WM8960 soundcard service.
```