waveshareteam/WM8960-Audio-HAT #63: WM8960-Soundcard Service Failure on Raspberry Pi with Kernel 6.6.47

jonathanpiette, 2024-09-26:
## Platform Information:

```
Raspberry Pi Model: Pi Zero W 2 and Pi 4
Kernel Version: 6.6.47+rpt-rpi-v8
Operating System: Debian-based Raspberry Pi OS
WM8960 HAT Repository: https://github.com/waveshare/WM8960-Audio-HAT
```
## Problem Description:

The wm8960-soundcard.service fails to start on both the Raspberry Pi Zero W 2 and Raspberry Pi 4 after following the installation procedure for the WM8960 HAT. The soundcard is detected, and both playback and recording functionalities are working, but the service fails with exit code 99. 

## Expected Behavior:

The wm8960-soundcard.service should successfully load the device tree overlay and configure ALSA to recognize and initialize the WM8960 HAT soundcard.
The soundcard should be automatically initialized and work on boot without manual intervention.

## Actual Behavior:

The wm8960-soundcard.service fails with exit code 99.
Device tree overlay application fails within the service, though it works when manually applied via ```/boot/firmware/config.txt```.
Manual ALSA configuration is required for soundcard initialization.

## Steps Taken to Investigate and Resolve the Issue:
### Step 1: Installation Procedure

We followed the instructions in the official [WM8960 HAT repository](https://github.com/waveshare/WM8960-Audio-HAT) as follows:

```bash
Copy code
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install git
git clone https://github.com/waveshare/WM8960-Audio-HAT
cd WM8960-Audio-HAT
sudo ./install.sh
sudo reboot
```
After reboot, the wm8960-soundcard.service failed to start.

### Step 2: Checking Service Status

The service status consistently failed with the following logs:

```bash
sudo systemctl status wm8960-soundcard.service

× wm8960-soundcard.service - WM8960 soundcard service
     Loaded: loaded (/lib/systemd/system/wm8960-soundcard.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since [timestamp]; 
    Process: [pid] ExecStart=/usr/bin/wm8960-soundcard (code=exited, status=99)
   Main PID: [pid] (code=exited, status=99)
```
journalctl -xeu wm8960-soundcard.service
This revealed that the device tree overlay failed to apply:

```bash
* Failed to apply overlay 'wm8960-soundcard' (kernel)
```
### Step 3: Investigating Overlay Loading

We manually checked the device tree overlays using:

```bash
Copy code
sudo dtoverlay -l
```
Output:

```bash
0:  dtparam i2c_arm=on
1:  wm8960-soundcard
```
This shows that the overlay is loaded manually via ```/boot/firmware/config.txt```.

We confirmed that the following line was present in ```/boot/firmware/config.txt```:

```bash
dtoverlay=wm8960-soundcard
```
### Step 4: Manual ALSA Configuration

Since the service failed to configure ALSA, we manually created the required symlinks and saved the ALSA state:

```bash
sudo alsactl restore
```
Playback and recording functionality were confirmed by running:

```bash
aplay -l
arecord -l
```
Output:
```bash
**** List of PLAYBACK Hardware Devices ****
card 3: wm8960soundcard [wm8960-soundcard], device 0: fe203000.i2s-wm8960-hifi wm8960-hifi-0 [fe203000.i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
...
**** List of CAPTURE Hardware Devices ****
card 3: wm8960soundcard [wm8960-soundcard], device 0: fe203000.i2s-wm8960-hifi wm8960-hifi-0 [fe203000.i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
Testing the recording and playback:
```bash
arecord -D hw:3,0 -f cd test.wav
aplay test.wav
```

Both recording and playback worked without issues.

### Step 5: Fixing Mono Channel Playback Issues

When attempting to play a mono .wav file (Front_Center.wav), we encountered a "Channels count not available" error:

```bash
aplay -D hw:3,0 /usr/share/sounds/alsa/Front_Center.wav
aplay: set_params:1358: Channels count non available
```

We fixed this by converting the file to stereo:

```bash
sudo apt-get install sox
sox /usr/share/sounds/alsa/Front_Center.wav -c 2 /tmp/Front_Center_stereo.wav
aplay -D hw:3,0 /tmp/Front_Center_stereo.wav
```
This resolved the issue.

### Step 6: Disabling the Failing Service

Since the soundcard worked manually, we decided to disable the failing wm8960-soundcard.service:

```bash
sudo systemctl disable wm8960-soundcard.service
```
We relied on manual setup (ALSA configuration and /boot/firmware/config.txt) to manage the soundcard, which worked consistently after reboots.

Final Outcome and Observations:
Manual Configuration Works: After manually applying the device tree overlay and configuring ALSA, the WM8960 soundcard works correctly for both playback and recording.
Service Failure: The wm8960-soundcard.service consistently fails to apply the device tree overlay, resulting in exit code 99.

ABelliqueux, 2024-12-27:
These step-by-step instructions look great, although in **Step 4: Manual ALSA Configuration**, you say : 
>  we manually created the required symlinks 

Could you elaborate on that step so that I can replicate your setup ?

Thank you !

TimH-droid, 2025-02-13:
I have the same question as @jonathanpiette .

In addition - although the line dtoverlay=wm8960-soundcard was indeed present in the config.sys text in my case the soundcard was not loaded as overlay 1

thanks for any help with this.  stuck at the moment

ForbiddenEra, 2026-01-04:
I'm also curious about these 'symlinks' ?

edit: oh.. the script for the service itself has it: @ABelliqueux if you still curious
```
$ cat /usr/bin/wm8960-soundcard
#!/bin/bash

set -x
exec 1>/var/log/$(basename $0).log 2>&1
#enable i2c interface
dtparam i2c_arm=on
modprobe i2c-dev

sleep 5
for loop in 1 2 3 4 5
do
    is_1a=$(i2cdetect -y  1 0x1a 0x1a | egrep "(1a|UU)" | awk '{print $2}')

    if [ "x${is_1a}" != "x" ] ; then
        echo "install wm8960-soundcard"
        dtoverlay wm8960-soundcard
        sleep 1

        rm /etc/asound.conf
        rm /var/lib/alsa/asound.state

        echo "create wm8960-soundcard configure file"
        ln -s /etc/wm8960-soundcard/asound.conf /etc/asound.conf

        echo "create wm8960-soundcard status file"
        ln -s /etc/wm8960-soundcard/wm8960_asound.state /var/lib/alsa/asound.state
        break
    fi
    sleep 3
done

alsactl restore
```