waveshareteam/WM8960-Audio-HAT #66: Will this ever upstream?

geiseri, 2025-01-06:
Is there a technical reason that this driver cannot be pushed upstream?  It is quite troublesome to use on lower end systems that do not have dkms or kernel headers.

medoc92, 2025-01-08:
Actually it is already in recent kernels, but on an rpi you may need an rpi-update to make it work. I wished they updated the doc which comes with the device, I lost a lot of time with this.

y-nk, 2025-04-01:
@medoc92 can you elaborate/explain how you did?

medoc92, 2025-04-01:
I already don't quite remember, but I really think that I just ran rpi-update to update the firmware, and things just got working.  There definitely is a wm8960 driver in 5 series and 6 series kernels.

medoc92, 2025-04-01:
And also have a look at my somewhat cryptic comment here: https://github.com/waveshareteam/WM8960-Audio-HAT/issues/67

Apparently I had to edit /boot/config.txt and do what the install.sh script from this driver does, which is apparently (from a diff of my config.txt to a standard one):

````
42,43c42,43
< #dtparam=i2c_arm=on
< #dtparam=i2s=on
---
> dtparam=i2c_arm=on
> dtparam=i2s=on
51d50
< 
54c53,54
< 
---
> dtoverlay=wm8960-soundcard
> dtoverlay=i2s-mmap
````

y-nk, 2025-04-02:
thanks a ton for the prompt answer @medoc92 . really appreciated!

medoc92, 2025-04-02:
Thanks but dit it work ?