waveshareteam/WM8960-Audio-HAT #68: WM8960-Soundcard Service Failure on Raspberry Pi with Kernel 6.12.2

zytegalaxy, 2025-04-01:
Raspberry Pi has release the new kernel version `6.12.2` (a bump from `6.6.74`) in March 2025. This has broken the WM8960 audio / sound card. 

I get the following error when running aplay: `unable to install hw params`

[image]

AndyW999, 2025-04-02:
I have submitted a request via tech support for this.

[WM8960 HAT.txt](https://github.com/user-attachments/files/19562791/WM8960.HAT.txt)

zytegalaxy, 2025-04-02:
@AndyW999 Thanks for the quick response on this. I have been investigating this a bit further. I was able to get the audio working on kernel `6.12.20` by installing seeed studio driver from here: 

https://github.com/HinTak/seeed-voicecard/tree/v6.12

I tried to diff files between this repo and seeed driver repo to find out what has changed but was not able to pin point the change.

One thing to notice that I am getting `No MCLK configured` error in `dmeg` logs on `6.12.20`

----
I have another question. I noticed that Waveshare wm8960 soundcard already has its overlay in Raspberry Pi overlays and was wondering if this would mean we can avoid driver installation and use the codec in `sound/soc/codecs/wm8960.c`?

https://github.com/raspberrypi/firmware/tree/master/boot/overlays

```
Name:   wm8960-soundcard
Info:   Overlay for the Waveshare wm8960 soundcard
Load:   dtoverlay=wm8960-soundcard,<param>=<val>
Params: alsaname                Changes the card name in ALSA
        compatible              Changes the codec compatibility
```

AndyW999, 2025-04-02:
I just tried the Seeed one on a new Pi install and upgrade but it did not work for me - 

andy@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
andy@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 2: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
andy@raspberrypi:~ $ sudo arecord -f cd -Dhw:2 | aplay -Dhw:2
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
arecord: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 44100
PERIOD_TIME: (125011 125012)
PERIOD_SIZE: 5513
PERIOD_BYTES: 22052
PERIODS: 4
BUFFER_TIME: (500045 500046)
BUFFER_SIZE: 22052
BUFFER_BYTES: 88208
TICK_TIME: 0
aplay: read_header:2912: read error
andy@raspberrypi:~ $

zytegalaxy, 2025-04-02:
@AndyW999 Did you make sure to checkout `v6.12` branch from HinTak seeed repo? (`git checkout v6.12`)?

Also, I upgrade kernel first (`sudo apt upgrade`) before installing the driver.

AndyW999, 2025-04-02:
Did all that and also out the dts file in my config.txt but it did not work

andy@raspberrypi:~ $ uname -a
Linux raspberrypi 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux

andy@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
andy@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
andy@raspberrypi:~ $ sudo arecord -f cd -Dhw:0 | aplay -Dhw:0
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
arecord: set_params:1416: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 44100
PERIOD_TIME: (125011 125012)
PERIOD_SIZE: 5513
PERIOD_BYTES: 22052
PERIODS: 4
BUFFER_TIME: (500045 500046)
BUFFER_SIZE: 22052
BUFFER_BYTES: 88208
TICK_TIME: 0
aplay: read_header:2912: read error
andy@raspberrypi:~ $

AndyW999, 2025-04-02:
Possibly not.
I went to https://github.com/HinTak/seeed-voicecard/tree/v6.12 and downloaded the zip and used that - is this not correct?

zytegalaxy, 2025-04-02:
@AndyW999 please clone the repo (`git clone https://github.com/HinTak/seeed-voicecard`) and inside the repo run `git checkout v6.12` to make sure you are on that branch. I am not sure if the zip file will contain the branch code or the whole repo code. You can run `git branch` to make sure you are on the correct branch.

AndyW999, 2025-04-02:
Yes, it works!

But... this kernel update seems to have broken other things...

Click the Rpi Icon, then shutdown and the shutdown menu does not appear.

I also use GPOI3 for shutdown - 
dtoverlay=gpio-shutdown
And that has stopped working too!

I did try an apt-mark hold on the kernel but when I did an update it ignored it and updated the kernel anyway.

AndyW999, 2025-04-02:
Looks like there has been an update that fixes the Shutdown problems...

I wonder what else was not checked properly ;)

AndyW999, 2025-04-02:
I used the seeed-2mic-voicecard overlay.

My problem now is that the RPi volume control in the menu bar controls the speaker rather than the headphones - any suggestions?

elgringoYan, 2025-04-03:
I am having this same exact problem with Pi Zero 2W

zytegalaxy, 2025-04-03:
I fixed the issue. Seems like some API names have changed in Kernel 6.12.20 and `wm8960-soundcard.c` needed to get updated. I will create a PR momentarily and the maintainer will hopefully accept and merge it quickly

elgringoYan, 2025-04-03:
Great News !

Thank you

zytegalaxy, 2025-04-03:
I created the PR. I tested it on my end and works. It helps if you guys can also test and report back the result.

elgringoYan, 2025-04-03:
> I created the PR. I tested it on my end and works. It helps if you guys can also test and report back the result.

Thanks @zytegalaxy , i would like to help in testing, but could you please provide quick instructions on how i can pull the PR ?
I tried cloning the ubopod repository, but i guess that was not the right thing to do, , as i am having the same problem.

Thanks,

zytegalaxy, 2025-04-03:
@elgringoYan yes you can test the version on my repo since I have already merged my PR branch with the master branch on my repo.

hex0id, 2025-04-05:
@elgringoYan  @zytegalaxy 

Not sure what goes wrong here, but on exact same kernel as yours guys I cannot build it (using PR fork):

```
make -j4 KERNELRELEASE=6.12.20+rpt-rpi-v8 -C /lib/modules/6.12.20+rpt-rpi-v8/build M=/var/lib/dkms/wm8960-soundcard/1.0/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.12.20+rpt-rpi-v8 (aarch64)
```

Make log is full of errors.

zytegalaxy, 2025-04-05:
> @elgringoYan  @zytegalaxy
>
> Not sure what goes wrong here, but on exact same kernel as yours guys I cannot build it (using PR fork):
>
> ```
> make -j4 KERNELRELEASE=6.12.20+rpt-rpi-v8 -C [...]

Can you please include the full `make.log` here? Also on which Raspberry Pi are you running this?

hex0id, 2025-04-05:
> > [@elgringoYan](https://github.com/elgringoYan)  [@zytegalaxy](https://github.com/zytegalaxy)
> Can you please include the full `make.log` here? Also on which Raspberry Pi are you running this?

Raspberry Pi 4 Model B Rev 1.1hex0id@shabot:~/shabot/WM8960-Audio-HAT $

```sh
uname -a
Linux 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux
```

console:
```sh
git clone https://github.com/zytegalaxy/WM8960-Audio-HAT.git
cd WM8960-Audio-HAT
sudo ./install.sh
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1:1.20230405-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
dkms is already the newest version (3.0.10-8+deb12u1).
git is already the newest version (1:2.39.5-0+deb12u2).
i2c-tools is already the newest version (4.3-2+b3).
libasound2-plugins is already the newest version (1.2.7.1-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Error! The module/version combo: wm8960-soundcard-1.0 is not located in the DKMS tree.
Creating symlink /var/lib/dkms/wm8960-soundcard/1.0/source -> /usr/src/wm8960-soundcard-1.0
Sign command: /lib/modules/6.12.20+rpt-rpi-v8/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.12.20+rpt-rpi-v8 -C /lib/modules/6.12.20+rpt-rpi-v8/build M=/var/lib/dkms/wm8960-soundcard/1.0/build......(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.12.20+rpt-rpi-v8 (aarch64)
Consult /var/lib/dkms/wm8960-soundcard/1.0/build/make.log for more information.
Created symlink /etc/systemd/system/sysinit.target.wants/wm8960-soundcard.service → /lib/systemd/system/wm8960-soundcard.service.
Job for wm8960-soundcard.service failed because the control process exited with error code.
See "systemctl status wm8960-soundcard.service" and "journalctl -xeu wm8960-soundcard.service" for details.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
```

make log:
```
make: Entering directory '/usr/src/linux-headers-6.12.20+rpt-rpi-v8'
  CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.o
  CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.o
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:26:27: error: ‘asoc_simple_startup’ undeclared here (not in a function)
   26 |         .startup        = asoc_simple_startup,
      |                           ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:27:27: error: ‘asoc_simple_shutdown’ undeclared here (not in a function)
   27 |         .shutdown       = asoc_simple_shutdown,
      |                           ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:28:27: error: ‘asoc_simple_hw_params’ undeclared here (not in a function)
   28 |         .hw_params      = asoc_simple_hw_params,
      |                           ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:106:41: warning: ‘struct asoc_simple_data’ declared inside parameter list will not be visible outside of this definition or declaration
[... 532 lines cut ...]
  610 |                                     simple_count_noml,
      |                                     ^~~~~~~~~~~~~~~~~
      |                                     |
      |                                     int (*)(struct asoc_simple_priv *, struct device_node *, struct device_node *, struct link_info *, bool) {aka int (*)(struct asoc_simple_priv *, struct device_node *, str
uct device_node *, struct link_info *, _Bool)}
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:436:39: note: expected ‘int (*)(struct asoc_simple_priv *, struct device_node *, struct device_node *, struct link_info *, bool)’ {aka ‘int (*)(struct aso
c_simple_priv *, struct device_node *, struct device_node *, struct link_info *, _Bool)’} but argument is of type ‘int (*)(struct asoc_simple_priv *, struct device_node *, struct device_node *, struct link_info *,
bool)’ {aka ‘int (*)(struct asoc_simple_priv *, struct device_node *, struct device_node *, struct link_info *, _Bool)’}
  436 |                                 int (*func_noml)(struct asoc_simple_priv *priv,
      |                                 ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  437 |                                                  struct device_node *np,
      |                                                  ~~~~~~~~~~~~~~~~~~~~~~~
  438 |                                                  struct device_node *codec,
      |                                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~
  439 |                                                  struct link_info *li, bool is_top),
      |                                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
      |                                     int (*)(struct asoc_simple_priv *, struct device_node *, struct device_node *, struct link_info *, bool) {aka int (*)(struct asoc_simple_priv *, struct device_node *, str
uct device_node *, struct link_info *, _Bool)}
/var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:440:39: note: expected ‘int (*)(struct asoc_simple_priv *, struct device_node *, struct device_node *, struct link_info *, bool)’ {aka ‘int (*)(struct aso
c_simple_priv *, struct device_node *, struct device_node *, struct link_info *, _Bool)’} but argument is of type ‘int (*)(struct asoc_simple_priv *, struct device_node *, struct device_node *, struct link_info *, :
```

zytegalaxy, 2025-04-05:
You are cloning from the wrong URL. Please clone from 

https://github.com/ubopod/WM8960-Audio-HAT.git

hex0id, 2025-04-05:
> You are cloning from the wrong URL. Please clone from
> 
> https://github.com/ubopod/WM8960-Audio-HAT.git
> 
> Make sure you run `./uninstall.sh` before installing again.

I actually did from both, but I'll try again.

miss-sammie, 2025-04-08:
hii, I'm having the same issue on a pi zero 2 W. 

I installed the stock waveshare driver and the seeed studio driver before reading to the end of the thread, but I did uninstall them both first before installing from your source. 

Still got these errors at the end: 
`Job for wm8960-soundcard.service failed because the control process exited with error code.
See "systemctl status wm8960-soundcard.service" and "journalctl -xeu wm8960-soundcard.service" for details.`

full logs below
`whisper@dream-machine:~/WM8960-Audio-HAT $ cd ..
whisper@dream-machine:~ $ git clone https://github.com/ubopod/WM8960-Audio-HAT.git
fatal: destination path 'WM8960-Audio-HAT' already exists and is not an empty directory.
whisper@dream-machine:~ $ ls
dream-machine  seeed-voicecard  WM8960-Audio-HAT
whisper@dream-machine:~ $ git clone https://github.com/ubopod/WM8960-Audio-HAT.git
Cloning into 'WM8960-Audio-HAT'...
remote: Enumerating objects: 93, done.
remote: Counting objects: 100% (56/56), done.
remote: Compressing objects: 100% (28/28), done.
remote: Total 93 (delta 41), reused 28 (delta 28), pack-reused 37 (from 1)
Receiving objects: 100% (93/93), 66.18 KiB | 752.00 KiB/s, done.
Resolving deltas: 100% (48/48), done.
whisper@dream-machine:~ $ cd WM8960-Audio-HAT
whisper@dream-machine:~/WM8960-Audio-HAT $ sudo ./install
sudo: ./install: command not found
whisper@dream-machine:~/WM8960-Audio-HAT $ sudo ./install.sh
Hit:1 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:2 http://raspbian.raspberrypi.com/raspbian bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
11 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1:1.20230405-1).
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up raspberrypi-kernel (1:1.20230405-1) ...
run-parts: executing /etc/kernel/postinst.d/dkms 6.1.21+ /boot/kernel.img
/etc/kernel/postinst.d/dkms:
dkms: running auto installation service for kernel 6.1.21+.
dkms: autoinstall for kernel: 6.1.21+.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 6.1.21+ /boot/kernel.img
run-parts: executing /etc/kernel/postinst.d/z50-raspi-firmware 6.1.21+ /boot/kernel.img
run-parts: executing /etc/kernel/postinst.d/dkms 6.1.21-v7+ /boot/kernel7.img
/etc/kernel/postinst.d/dkms:
dkms: running auto installation service for kernel 6.1.21-v7+.
dkms: autoinstall for kernel: 6.1.21-v7+.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 6.1.21-v7+ /boot/kernel7.img
run-parts: executing /etc/kernel/postinst.d/z50-raspi-firmware 6.1.21-v7+ /boot/kernel7.img
run-parts: executing /etc/kernel/postinst.d/dkms 6.1.21-v7l+ /boot/kernel7l.img
/etc/kernel/postinst.d/dkms:
dkms: running auto installation service for kernel 6.1.21-v7l+.
dkms: autoinstall for kernel: 6.1.21-v7l+.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 6.1.21-v7l+ /boot/kernel7l.img
run-parts: executing /etc/kernel/postinst.d/z50-raspi-firmware 6.1.21-v7l+ /boot/kernel7l.img
run-parts: executing /etc/kernel/postinst.d/dkms 6.1.21-v8+ /boot/kernel8.img
/etc/kernel/postinst.d/dkms:
dkms: autoinstall for kernel 6.1.21-v8+ was skipped since the kernel headers for this kernel do not seem to be installed.
run-parts: executing /etc/kernel/postinst.d/initramfs-tools 6.1.21-v8+ /boot/kernel8.img
run-parts: executing /etc/kernel/postinst.d/z50-raspi-firmware 6.1.21-v8+ /boot/kernel8.img
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
dkms is already the newest version (3.0.10-8+deb12u1).
git is already the newest version (1:2.39.5-0+deb12u2).
i2c-tools is already the newest version (4.3-2+b2).
libasound2-plugins is already the newest version (1.2.7.1-1).
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.
Error! The module/version combo: wm8960-soundcard-1.0 is not located in the DKMS tree.
Creating symlink /var/lib/dkms/wm8960-soundcard/1.0/source -> /usr/src/wm8960-soundcard-1.0
Sign command: /lib/modules/6.6.51+rpt-rpi-v7/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.6.51+rpt-rpi-v7 -C /lib/modules/6.6.51+rpt-rpi-v7/build M=/var/lib/dkms/wm8960-soundcard/1.0/build.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.6.51+rpt-rpi-v7 (armv7l)
Consult /var/lib/dkms/wm8960-soundcard/1.0/build/make.log for more information.
Created symlink /etc/systemd/system/sysinit.target.wants/wm8960-soundcard.service → /lib/systemd/system/wm8960-soundcard.service.
Job for wm8960-soundcard.service failed because the control process exited with error code.
See "systemctl status wm8960-soundcard.service" and "journalctl -xeu wm8960-soundcard.service" for details.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!`

zytegalaxy, 2025-04-09:
@miss-sammie It seems like from the logs that the driver is being built with older version of the kernel:

```
make -j4 KERNELRELEASE=6.6.51+rpt-rpi-v7 -C /lib/modules/6.6.51+rpt-rpi-v7/build M=/var/lib/dkms/wm8960-soundcard/1.0/build.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.6.51+rpt-rpi-v7 (armv7l)
```

Can you please run `uname -r` and check your kernel version? I think my changes are backward incompatible with older kernel versions and kernel version must be 6.12 and more recent for a successful install.

It might be that `upgrade` did not succeed in your case and kernel is still on the older version. Please make sure to reboot after upgrade and check your kernel version before installing the driver from the repo.

miss-sammie, 2025-04-09:
Ahh ok! So yes I did need to restart the pi after performing the upgrade, and the kernel is showing as 6.12 after upgrade. I did get errors when I installed the driver, but it works!! 

Here are the logs in case it's helpful. 

Thank you!

```
whisper@dream-machine:~/WM8960-Audio-HAT $ sudo ./install.sh
Hit:1 http://raspbian.raspberrypi.com/raspbian bookworm InRelease
Hit:2 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1:1.20230405-1).
The following package was automatically installed and is no longer required:
  libcamera0.3
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
dkms is already the newest version (3.0.10-8+deb12u1).
git is already the newest version (1:2.39.5-0+deb12u2).
i2c-tools is already the newest version (4.3-2+b2).
libasound2-plugins is already the newest version (1.2.7.1-1).
The following package was automatically installed and is no longer required:
  libcamera0.3
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Error! The module/version combo: wm8960-soundcard-1.0 is not located in the DKMS tree.
Creating symlink /var/lib/dkms/wm8960-soundcard/1.0/source -> /usr/src/wm8960-soundcard-1.0
Sign command: /lib/modules/6.12.20+rpt-rpi-v7/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.12.20+rpt-rpi-v7 -C /lib/modules/6.12.20+rpt-rpi-v7/build M=/var/lib/dkms/wm8960-soundcard/1.0/build......
Signing module /var/lib/dkms/wm8960-soundcard/1.0/build/snd-soc-wm8960.ko
Signing module /var/lib/dkms/wm8960-soundcard/1.0/build/snd-soc-wm8960-soundcard.ko
Cleaning build area...

snd-soc-wm8960.ko.xz:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/6.12.20+rpt-rpi-v7/updates/dkms/

snd-soc-wm8960-soundcard.ko.xz:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/6.12.20+rpt-rpi-v7/updates/dkms/
depmod......
Created symlink /etc/systemd/system/sysinit.target.wants/wm8960-soundcard.service → /lib/systemd/system/wm8960-soundcard.service.
Job for wm8960-soundcard.service failed because the control process exited with error code.
See "systemctl status wm8960-soundcard.service" and "journalctl -xeu wm8960-soundcard.service" for details.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
```

boivinx7, 2025-04-26:
So i was getting the errors, thats why im here of course, but now, everything installed fine aka no errors, but  
`aplay -l` 
and
`arecord -l`
no longer outputs me the wm8960-soundcard in the list

boivinx7, 2025-04-26:
So i reinstalled it all, and seems that i loose the devices on reboot...

boivinx7, 2025-04-26:
Ok so ive put the new overlay, then instead of `sudo reboot` i unplugged the pi and plugged it back in, not both devices show. So weird

rubenlohaus, 2025-05-31:
> e

try:

`sudo modprobe snd-soc-wm8960`

rubenlohaus, 2025-05-31:
@boivinx7 try:

`sudo modprobe snd-soc-wm8960`

then 

`aplay -l`

zytegalaxy, 2025-07-16:
@erickmob it works with pi 5. Does the card get recognized by `aplay -l`?

zytegalaxy, 2025-07-17:
@erickmob If it is detected by `aplay` then it should work. Did you try playing a sample wave file with aplay and see what happens? 

yes for installing all you need to do is to run `instal.sh` script. It should take care of everything. Make sure you restart after installing the drivers. 

Alternatively you can try the pre-built image that has the driver installed:

https://github.com/ubopod/ubo_app/releases/download/v1.5.0/ubo_app-1.5.0-bookworm-lite.img.gz

This is the image for the open source ai assistant I am working on ([ubo app](https://github.com/ubopod/ubo_app)). So it has a bunch of other things installed. At least this way you can rule out hardware issues.

erickmob, 2025-07-18:
thx for replying @zytegalaxy 

Ok, here's what i did:
- clean install of PI OS 64-bit;
- clone repo and run install script.

```
186 packages can be upgraded. Run 'apt list --upgradable' to see them.
N: Repository 'http://deb.debian.org/debian bookworm InRelease' changed its 'Version' value from '12.10' to '12.11'
Created symlink /etc/systemd/system/sysinit.target.wants/wm8960-soundcard.service → /lib/systemd/system/wm8960-soundcard.service.
Job for wm8960-soundcard.service failed because the control process exited with error code.
See "systemctl status wm8960-soundcard.service" and "journalctl -xeu wm8960-soundcard.service" for details.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------

``

- reboot pi with `sudo reboot`

```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: wm8960soundcard [wm8960-soundcard], device 0: 1f000a0000.i2s-wm8960-hifi wm8960-hifi-0 [1f000a0000.i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

Then tested with: 
```
pi@raspberrypi:~ $ sudo arecord -f cd -Dhw:0 | aplay -Dhw:0
aplay: main:831: audio open error: Unknown error 524
arecord: main:831: audio open error: No such file or directory
```

Not sure what i'm missing here. 

I'll try that `pre-built image` later and post the results here as well

barkside, 2025-08-16:
I managed to get mine installed and recognised on latest everything - took a while!  But the sound is not working - just the occasional popping noise/sound when I play test files.  No errors, apart from dmesg: "Error: Driver 'asoc-simple-card' is already registered, aborting..."

Any ideas?

RASPIAUDIOadmin, 2025-08-22:
same issue here with Linux raspberrypi 6.12.42-v8-16k+ #1899 SMP PREEMPT Tue Aug 19 15:13:04 BST 2025 aarch64 GNU/Linux
using dtoverlay=wm8960-soundcard.dtbo

Detected and I can change controls in alsamixer but no sound coming out

I'm using the Raspiaudio ULTRA that was working with older kernel

zytegalaxy, 2025-08-22:
It seems like that the device tree objects are already included in the latest Kernel version and it is causing some conflicts. I suspect that it might be possible to use this sound card without the need to install the drive going forward. I have not had time to investigate this fully though.