waveshareteam/WM8960-Audio-HAT #69: Don't work with Kernel 6.12.20+rpt-rpi-v7

DG1FI, 2025-04-02:
During install:

> Creating symlink /var/lib/dkms/wm8960-soundcard/1.0/source -> /usr/src/wm8960-soundcard-1.0
> Sign command: /lib/modules/6.12.20+rpt-rpi-v7/build/scripts/sign-file
> Signing key: /var/lib/dkms/mok.key
> Public certificate (MOK): /var/lib/dkms/mok.pub
> 
> Building module:
> Cleaning build area...
> make -j4 KERNELRELEASE=6.12.20+rpt-rpi-v7 -C /lib/modules/6.12.20+rpt-rpi-v7/build M=/var/lib/dkms/wm8960-soundcard/1.0/build......(bad exit status: 2)
> Error! Bad return status for module build on kernel: 6.12.20+rpt-rpi-v7 (armv7l)
> Consult /var/lib/dkms/wm8960-soundcard/1.0/build/make.log for more information.

make.log

> DKMS make.log for wm8960-soundcard-1.0 for kernel 6.12.20+rpt-rpi-v7 (armv7l)
> Mi 2. Apr 20:36:01 CEST 2025
> make: Verzeichnis „/usr/src/linux-headers-6.12.20+rpt-rpi-v7“ wird betreten
>   CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960.o
>   CC [M]  /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.o
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:26:27: error: ‘asoc_simple_startup’ undeclared here (not in a function)
>    26 |         .startup        = asoc_simple_startup,
>       |                           ^~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:27:27: error: ‘asoc_simple_shutdown’ undeclared here (not in a function)
>    27 |         .shutdown       = asoc_simple_shutdown,
>       |                           ^~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:28:27: error: ‘asoc_simple_hw_params’ undeclared here (not in a function)
>    28 |         .hw_params      = asoc_simple_hw_params,
>       |                           ^~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:106:41: warning: ‘struct asoc_simple_data’ declared inside parameter list will not be visible outside of this definition or declaration
>   106 |                                  struct asoc_simple_data *adata)
>       |                                         ^~~~~~~~~~~~~~~~
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c: In function ‘simple_parse_convert’:
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:111:9: error: implicit declaration of function ‘asoc_simple_parse_convert’; did you mean ‘simple_parse_convert’? [-Werror=implicit-function-declaration]
>   111 |         asoc_simple_parse_convert(top,  PREFIX, adata);
>       |         ^~~~~~~~~~~~~~~~~~~~~~~~~
>       |         simple_parse_convert
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c: At top level:
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:137:37: warning: ‘struct asoc_simple_priv’ declared inside parameter list will not be visible outside of this definition or declaration
>   137 | static int simple_parse_node(struct asoc_simple_priv *priv,
>       |                                     ^~~~~~~~~~~~~~~~
> In file included from /usr/src/linux-headers-6.12.20+rpt-common-rpi/include/sound/simple_card.h:13,
>                  from /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:15:
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c: In function ‘simple_parse_node’:
> /usr/src/linux-headers-6.12.20+rpt-common-rpi/include/sound/simple_card_utils.h:79:49: error: invalid use of undefined type ‘struct asoc_simple_priv’
>    79 | #define simple_priv_to_card(priv)       (&(priv)->snd_card)
>       |                                                 ^~
> /usr/src/linux-headers-6.12.20+rpt-common-rpi/include/sound/simple_card_utils.h:81:42: note: in expansion of macro ‘simple_priv_to_card’
>    81 | #define simple_priv_to_dev(priv)        (simple_priv_to_card(priv)->dev)
>       |                                          ^~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:143:30: note: in expansion of macro ‘simple_priv_to_dev’
>   143 |         struct device *dev = simple_priv_to_dev(priv);
>       |                              ^~~~~~~~~~~~~~~~~~
> /usr/src/linux-headers-6.12.20+rpt-common-rpi/include/sound/simple_card_utils.h:79:49: error: invalid use of undefined type ‘struct asoc_simple_priv’
>    79 | #define simple_priv_to_card(priv)       (&(priv)->snd_card)
>       |                                                 ^~
> /usr/src/linux-headers-6.12.20+rpt-common-rpi/include/sound/simple_card_utils.h:82:42: note: in expansion of macro ‘simple_priv_to_card’
>    82 | #define simple_priv_to_link(priv, i)    (simple_priv_to_card(priv)->dai_link + (i))
>       |                                          ^~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:145:45: note: in expansion of macro ‘simple_priv_to_link’
>   145 |         struct snd_soc_dai_link *dai_link = simple_priv_to_link(priv, li->link);
>       |                                             ^~~~~~~~~~~~~~~~~~~
> /usr/src/linux-headers-6.12.20+rpt-common-rpi/include/sound/simple_card_utils.h:80:48: error: invalid use of undefined type ‘struct asoc_simple_priv’
>    80 | #define simple_priv_to_props(priv, i)   ((priv)->dai_props + (i))
>       |                                                ^~
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:146:46: note: in expansion of macro ‘simple_priv_to_props’
>   146 |         struct simple_dai_props *dai_props = simple_priv_to_props(priv, li->link);
>       |                                              ^~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:152:23: error: implicit declaration of function ‘asoc_link_to_cpu’; did you mean ‘snd_soc_link_to_cpu’? [-Werror=implicit-function-declaration]
>   152 |                 dlc = asoc_link_to_cpu(dai_link, 0);
[... 501 lines cut ...]
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:712:50: error: invalid use of undefined type ‘struct asoc_simple_card_info’
>   712 |                 memcpy(dai_props->cpu_dai, &cinfo->cpu_dai,
>       |                                                  ^~
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:714:52: error: invalid use of undefined type ‘struct asoc_simple_card_info’
>   714 |                 memcpy(dai_props->codec_dai, &cinfo->codec_dai,
>       |                                                    ^~
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:720:9: error: implicit declaration of function ‘asoc_simple_debug_info’ [-Werror=implicit-function-declaration]
>   720 |         asoc_simple_debug_info(priv);
>       |         ^~~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:729:9: error: implicit declaration of function ‘asoc_simple_clean_reference’ [-Werror=implicit-function-declaration]
>   729 |         asoc_simple_clean_reference(card);
>       |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c: At top level:
> /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.c:749:19: error: ‘asoc_simple_remove’ undeclared here (not in a function); did you mean ‘asoc_simple_probe’?
>   749 |         .remove = asoc_simple_remove,
>       |                   ^~~~~~~~~~~~~~~~~~
>       |                   asoc_simple_probe
> cc1: some warnings being treated as errors
> make[2]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/scripts/Makefile.build:249: /var/lib/dkms/wm8960-soundcard/1.0/build/wm8960-soundcard.o] Fehler 1
> make[1]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:1969: /var/lib/dkms/wm8960-soundcard/1.0/build] Fehler 2
> make: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:236: __sub-make] Fehler 2
> make: Verzeichnis „/usr/src/linux-headers-6.12.20+rpt-rpi-v7“ wird verlassen
> 
> 
>

zytegalaxy, 2025-04-02:
duplicate of https://github.com/waveshareteam/WM8960-Audio-HAT/issues/68

DG1FI, 2025-04-03:
Yes - but an newer Kernel version

zytegalaxy, 2025-04-03:
@DG1FI Please checkout my PR. It has addressed this issue.

DG1FI, 2025-04-07:
@zytegalaxy your branch works good with the new kernel. seems there is an overlay for raspberry pi? but not found any informations about this.

zytegalaxy, 2025-04-14:
@DG1FI I see this overlay being added to https://github.com/raspberrypi/firmware/tree/master/boot/overlays

```
Name:   wm8960-soundcard
Info:   Overlay for the Waveshare wm8960 soundcard
Load:   dtoverlay=wm8960-soundcard,<param>=<val>
Params: alsaname                Changes the card name in ALSA
        compatible              Changes the codec compatibility
```

but not sure if it would work with latest kernel version. have not done any testing on it...