waveshareteam/WM8960-Audio-HAT #76: Driver not working for RP5 and Linux raspberrypi 6.12.25+rpt-rpi-2712

erickmob, 2025-07-16:
I don't know what else to do to make this work. 

I've already: 
-  followed  [waveshare FAQ ](https://www.waveshare.com/wiki/WM8960_Audio_HAT#FAQ) and set `arm_64bit=0` to config file. 
- [piRadio](https://github.com/bobrathbone/piradio6/blob/master/scripts/install_wm8960.sh) script
- [ubopod](https://github.com/waveshareteam/WM8960-Audio-HAT/issues/74) script as recommended from this [issue](https://github.com/waveshareteam/WM8960-Audio-HAT/issues/74) and this [other issue](https://github.com/waveshareteam/WM8960-Audio-HAT/issues/51) 
- HinTak driver

Nothing ever works. 
```
pi@raspberrypi:~ $ git clone https://github.com/waveshareteam/WM8960-Audio-HAT.git
Cloning into 'WM8960-Audio-HAT'...
remote: Enumerating objects: 98, done.
remote: Counting objects: 100% (61/61), done.
remote: Compressing objects: 100% (29/29), done.
remote: Total 98 (delta 45), reused 34 (delta 32), pack-reused 37 (from 1)
Receiving objects: 100% (98/98), 66.87 KiB | 6.08 MiB/s, done.
Resolving deltas: 100% (52/52), done.
pi@raspberrypi:~ $ cd WM8960-Audio-HAT/
pi@raspberrypi:~/WM8960-Audio-HAT $ sudo ./install.sh 
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
186 packages can be upgraded. Run 'apt list --upgradable' to see them.
Created symlink /etc/systemd/system/sysinit.target.wants/wm8960-soundcard.service → /lib/systemd/system/wm8960-soundcard.service.
Job for wm8960-soundcard.service failed because the control process exited with error code.
See "systemctl status wm8960-soundcard.service" and "journalctl -xeu wm8960-soundcard.service" for details.
------------------------------------------------------
Please reboot your raspberry pi to apply all settings
Enjoy!
------------------------------------------------------
pi@raspberrypi:~/WM8960-Audio-HAT $ sudo nano /boot/firmware/config.txt
pi@raspberrypi:~/WM8960-Audio-HAT $ systemctl status wm8960-soundcard.service
× wm8960-soundcard.service - WM8960 soundcard service
     Loaded: loaded (/lib/systemd/system/wm8960-soundcard.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Wed 2025-07-16 18:03:24 BST; 33min ago
    Process: 2417 ExecStart=/usr/bin/wm8960-soundcard (code=exited, status=99)
   Main PID: 2417 (code=exited, status=99)
        CPU: 100ms

Jul 16 18:03:18 raspberrypi systemd[1]: Starting wm8960-soundcard.service - WM8960 soundcard service...
Jul 16 18:03:18 raspberrypi wm8960-soundcard[2417]: + exec
Jul 16 18:03:18 raspberrypi wm8960-soundcard[2418]: ++ basename /usr/bin/wm8960-soundcard
Jul 16 18:03:24 raspberrypi systemd[1]: wm8960-soundcard.service: Main process exited, code=exited, st>
Jul 16 18:03:24 raspberrypi systemd[1]: wm8960-soundcard.service: Failed with result 'exit-code'.
Jul 16 18:03:24 raspberrypi systemd[1]: Failed to start wm8960-soundcard.service - WM8960 soundcard se>

```

```
pi@raspberrypi:~/WM8960-Audio-HAT $ uname -a
Linux raspberrypi 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux
```

```
pi@raspberrypi:~/WM8960-Audio-HAT $ journalctl -xeu wm8960-soundcard.service

░░ A start job for unit wm8960-soundcard.service has begun execution.
░░ 
░░ The job identifier is 1449.
Jul 16 18:03:18 raspberrypi wm8960-soundcard[2417]: + exec
Jul 16 18:03:18 raspberrypi wm8960-soundcard[2418]: ++ basename /usr/bin/wm8960-soundcard
Jul 16 18:03:24 raspberrypi systemd[1]: wm8960-soundcard.service: Main process exited, code=exited, st>
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ An ExecStart= process belonging to unit wm8960-soundcard.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 99.
Jul 16 18:03:24 raspberrypi systemd[1]: wm8960-soundcard.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit wm8960-soundcard.service has entered the 'failed' state with result 'exit-code'.
Jul 16 18:03:24 raspberrypi systemd[1]: Failed to start wm8960-soundcard.service - WM8960 soundcard se>
░░ Subject: A start job for unit wm8960-soundcard.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit wm8960-soundcard.service has finished with a failure.
░░ 
░░ The job identifier is 1449 and the job result is failed.

```

What am I missing here?

erickmob, 2025-07-27:
Thx! Will try that later today

zerulin, 2025-09-28:
The batch 6.12x was merged to the master.

closed 2025-09-28 by zerulin